amzn/amzn-drivers #297: Ena memzones memory allocation issue with DPDK 23.11 ena driver

shrbandi, 2024-03-13:
### Preliminary Actions

- [X] I have searched the existing [issues](https://github.com/amzn/amzn-drivers/issues) and didn't find a duplicate.
- [X] I have followed the [AWS official troubleshoot documentation](https://docs.aws.amazon.com/AWSEC2/latest/UserGuide/troubleshooting-ena.html).
- [X] I have followed the driver [readme](https://github.com/amzn/amzn-drivers) and [best practices](https://github.com/amzn/amzn-drivers/blob/master/kernel/linux/ena/ENA_Linux_Best_Practices.rst).

### Driver Type

DPDK PMD for Elastic Network Adapter (ENA)

### Driver Tag/Commit

NA

### Custom Code

No

### OS Platform and Distribution

Linux 5.15.0 x86_64

### Support request

Hi,

In our application, after uplifting the DPDK version to 23.11 we have observed the unexpected behavior related to ENA mem zones, specifically ena_p0_mz0 and ena_p1_mz0, which are allocating a large amount of memory, measured in gigabytes (GB).

The function ena_com_allocate_customer_metrics_buffer() is newly introduced in DPDK 23.11 as part of the following patch:
https://inbox.dpdk.org/dev/20231026142135.19967-2-shaibran@amazon.com

In this patch, within the ena_com_allocate_customer_metrics_buffer() function, customer_metrics->buffer_len is initially set to zero to reserve memory for mem zones ena_p0_mz0 and ena_p1_mz0. Later, it is updated with the macro ENA_CUSTOMER_METRICS_BUFFER_SIZE (512 bytes).

We have noticed that when the size of the memzone is initially set to zero, ena_p0_mz0 and ena_p1_mz0 allocate a large amount of memory, measured in gigabytes, whereas this behavior was not observed prior i.e in dpdk 23.07, where these mem zones were allocated with only 64 bytes.

Is this behavior expected with the changes related to ena in dpdk 23.11? Additionally, is there a specific reason for updating buffer_len of customer metrics after reserving memory for memzones? Can't we assign buffer_len to ENA_CUSTOMER_METRICS_BUFFER_SIZE (512 bytes) before reserving memory?

Thanks
Shruthi

### Contact Details

shruthi.bandi@oracle.com

shaibran, 2024-03-27:
Hi Shruthi, and thank you for choosing ec2!
This is indeed a bug in a patch that reworked the mechanism that queries the performance metrics, which we recently fixed in DPDK 24.03-rc3.
We are preparing a fix that we intend to include it in the next stable release of DPDK 23.11 and will share the pre release patch with you once ready. 

All the best,
Shai

shrbandi, 2024-04-01:
Hi Shaibran,

Thank you for the clarification.

Regards,
Shruthi

shaibran, 2024-04-07:
[0001-net-ena-base-fix-metrics-excessive-memory-consump.patch](https://github.com/amzn/amzn-drivers/files/14896290/0001-net-ena-base-fix-metrics-excessive-memory-consump.patch)

closed 2024-04-07 by shaibran

shrbandi, 2024-04-08:
Hi @shaibran,

Acknowledged. Thanks for sharing the patch.

Regards,
Shruthi