amzn/amzn-drivers #298: [Feature request]: AF_XDP hardware timestamp support

wrfeewrtgaqwfrwaq, 2024-04-16:
### Preliminary Actions

- [X] I have searched the existing [issues](https://github.com/amzn/amzn-drivers/issues) and didn't find a duplicate.

### Driver Type

Linux kernel driver for Elastic Network Adapter (ENA)

### Motivation

Better time resolution for incoming packets when using userspace network stack like onload

### Solution

It'd be nice to support new `xmo_rx_timestamp()` API.
If not the at least userspace network timestamping API
https://www.kernel.org/doc/html/v5.15/networking/timestamping.html

### Contact Details

michal.pietrzak@radix-trading.com

davidarinzon, 2024-04-24:
Hi @wrfeewrtgaqwfrwaq 

Thank you for reaching out and raising this request.
We are looking into adding HW timestamping support to the ENA driver, please stay tuned for updates.
In the meanwhile, would appreciate to hear more about your use-cases which require better time resolution to be passed through AF_XDP.

selevit, 2025-06-20:
I see that the hardware timestamp support is added to the Linux ENA driver in [2.15.0.](https://github.com/amzn/amzn-drivers/releases/tag/ena_linux_2.15.0).

Looking forward for this to be available in the DPDK driver too!

closed 2025-12-28 by davidarinzon