amzn/amzn-drivers #300: [Support]: DPDK 21.11 onwards strange issues seen in AWS 'c5a' instance family

ganesh-patwardhan, 2024-05-07:
### Preliminary Actions

- [X] I have searched the existing [issues](https://github.com/amzn/amzn-drivers/issues) and didn't find a duplicate.
- [X] I have followed the [AWS official troubleshoot documentation](https://docs.aws.amazon.com/AWSEC2/latest/UserGuide/troubleshooting-ena.html).
- [X] I have followed the driver [readme](https://github.com/amzn/amzn-drivers) and [best practices](https://github.com/amzn/amzn-drivers/blob/master/kernel/linux/ena/ENA_Linux_Best_Practices.rst).

### Driver Type

DPDK PMD for Elastic Network Adapter (ENA)

### Driver Tag/Commit

ena_linux_2.0.2

### Custom Code

No

### OS Platform and Distribution

Custom linux 

### Support request

Hi With DPDK 21.11 onward (tried only DPDK 21.11 and 22.11) in AWS c5a instance family below issues are seen
- DHCP ip assigned to the  interface is getting lost after 4-5 hours
- Instance keeps sending DHCP request but not getting any response, same time rx_missed_errors are getting incremented for each DHCP packet
- This issue happens only in 'c5a' family (ex: c5a.4xlarge)
- When from external VM tcp/udp packets are sent to DPDK interface those are getting in to userspace when above mentioned issue is still present.
- When lot of TCP packets are sent to dpdk interface from outside suddenly things star to work (i.e interface gets DHCP IP)

Note: Its difficult to test with latest DPDK version in our product. please let me know if this is some known issue / fixed in latest version etc 

### Contact Details

gpatward@cisco.com

shaibran, 2024-05-08:
I am not aware of such issue, but we will try to reproduce in-house.
can you please share instance id and the time frame when this occurred? I will try looking in our ec2 logs

ganesh-patwardhan, 2024-05-08:
Thanks Shai

Instance id : i-0540d729ec7585563
Currently instance is in same state (instance has 4 interfaces and last 2 interfaces are using DPDK which are in problematic state currently)

Regards
Ganesh Patwardhan

shaibran, 2024-05-12:
Same issue is already tracked through internal ticket, resolving here.

closed 2024-05-12 by shaibran

shaibran, 2025-09-28:
To ensure transparency and facilitate resolution, we are providing a detailed description of the identified issue affecting DHCP packet reception, along with recommended mitigation steps.

**Observed behavior**:

- DHCP ingress packets cease arriving at the application after approximately 4–5 hours of operation.
- The `rx-missed` error counter increases in correlation with the missing DHCP packets. This indicates that packets are being dropped by the hardware due to unavailable RX buffers.
- Initiating ingress TCP traffic on the same RX queue temporarily restores DHCP packet flow to the application.

**Root cause:**
The RX queue refill threshold is dynamically calculated by the driver based on the configured queue depth. However, the application may override this threshold during setup, potentially configuring a value that prevents the RX queue from being properly refilled. This leads to buffer exhaustion and dropped DHCP packets.

**Recommended mitigation**:

When invoking `eth_dev_ops->rx_queue_setup`, the application must avoid overriding the driver's dynamic threshold calculation. Specifically, the `rx_free_thresh` field in `struct rte_eth_rxconf` should be set to one of the following values:
`0` Enables dynamic threshold calculation by the driver (recommended).
`32`Manually sets the threshold to 32, which has been validated to support DHCP traffic effectively.