amzn/amzn-drivers #302: [Bug]: Compilation of ena_linux v2.12.0 for kernels older than 6.8.0

kiike, 2024-05-14:
### Preliminary Actions

- [X] I have searched the existing [issues](https://github.com/amzn/amzn-drivers/issues) and didn't find a duplicate.
- [X] I have followed the [AWS official troubleshoot documentation](https://docs.aws.amazon.com/AWSEC2/latest/UserGuide/troubleshooting-ena.html).
- [X] I have followed the driver [readme](https://github.com/amzn/amzn-drivers) and [best practices](https://github.com/amzn/amzn-drivers/blob/master/kernel/linux/ena/ENA_Linux_Best_Practices.rst).

### Driver Type

Linux kernel driver for Elastic Network Adapter (ENA)

### Driver Tag/Commit

ena_linux-2.12.0

### Custom Code

No

### OS Platform and Distribution

```
╭──────────────────┬───────────────────────────────────────────────────────────╮
│ kernel-name      │ Linux                                                     │
│ nodename         │ builder                                                   │
│ kernel-release   │ 6.6.30                                                    │
│ kernel-version   │ #1-NixOS SMP PREEMPT_DYNAMIC Thu May  2 14:32:50 UTC 2024 │
│ machine          │ x86_64                                                    │
│ operating-system │ GNU/Linux                                                 │
╰──────────────────┴───────────────────────────────────────────────────────────╯
```

### Bug description

I am trying to package the ena Linux driver for several kernel versions as a NixOS package. 6.8 and 6.9 builds succeed perfectly thanks to the latests commits that introduce compatibility with such versions. However, when building for older kernels, it looks as if the `ethtool_puts` test is not being correct, and the build fails.
From the script results, the `ethtool_puts` is reported as available, but 6.6.30, for instance, doesn't have this API, so it should be reported as unavailable. 

### Reproduction steps

```bash
The build output in "actual behavior" contains the step by step commands and log.
```

### Expected Behavior

Clean build like for 6.8 or 6.9.

### Actual Behavior

<details>
<summary>build log for kernel 6.6.30</summary>
<code>
 nix build -L .#linuxPackages.ena
ena> Running phase: unpackPhase
ena> unpacking source archive /nix/store/7nnir1y4wikh13mrhrpdz0zk6ap9m4rn-source
ena> source root is source
ena> Running phase: patchPhase
ena> Running phase: updateAutotoolsGnuConfigScriptsPhase
ena> Running phase: configurePhase
ena> substituteStream(): WARNING: '--replace' is deprecated, use --replace-{fail,warn,quiet}. (file 'Makefile')
ena> Running phase: buildPhase
ena> build flags: SHELL=/nix/store/h3bhzvz9ipglcybbcvkxvm4vg9lwvqg4-bash-5.2p26/bin/bash O=\$\(buildRoot\) CC=/nix/store/md6hh4rrcrf99nssvcam3qaqs3skj086-gcc-wrapper-13.2.0/bin/cc HOSTCC=/nix/store/md6hh4rrcrf99nssvcam3qaqs3skj086-gcc-wrapper-13.2.0/bin/cc HOSTLD=/nix/store/m3gq2f6iwh3g67zrs27m7w82b37l8gf8-binutils-wrapper-2.41/bin/ld ARCH=x86_64
ena> ./configure.sh --kernel-dir /nix/store/w2bzm4kvp46xhsqpbqvp1yyxmp5zq0fz-linux-6.6.30-dev/lib/modules/6.6.30/build --verbose
ena> Trying to intercept CC arguments
ena> Detected compiler:  /nix/store/md6hh4rrcrf99nssvcam3qaqs3skj086-gcc-wrapper-13.2.0/bin/cc
ena> Managed to intercept CC arguments
ena> Command: /nix/store/md6hh4rrcrf99nssvcam3qaqs3skj086-gcc-wrapper-13.2.0/bin/cc -nostdinc -I/nix/store/w2bzm4kvp46xhsqpbqvp1yyxmp5zq0fz-linux-6.6.30-dev/lib/modules/6.6.30/source/arch/x86/include -I./arch/x86/include/generated -I/nix/store/w2bzm4kvp46xhsqpbqvp1yyxmp5zq0fz-linux-6.6.30-dev/lib/modules/6.6.30/source/include -I./include -I/nix/store/w2bzm4kvp46xhsqpbqvp1yyxmp5zq0fz-linux-6.6.30-dev/lib/modules/6.6.30/source/arch/x86/include/uapi -I./arch/x86/include/generated/uapi -I/nix/store/w2bzm4kvp46xhsqpbqvp1yyxmp5zq0fz-linux-6.6.30-dev/lib/modules/6.6.30/source/include/uapi -I./include/generated/uapi -include /nix/store/w2bzm4kvp46xhsqpbqvp1yyxmp5zq0fz-linux-6.6.30-dev/lib/modules/6.6.30/source/include/linux/compiler-version.h -include /nix/store/w2bzm4kvp46xhsqpbqvp1yyxmp5zq0fz-linux-6.6.30-dev/lib/modules/6.6.30/source/include/linux/kconfig.h -include /nix/store/w2bzm4kvp46xhsqpbqvp1yyxmp5zq0fz-linux-6.6.30-dev/lib/modules/6.6.30/source/include/linux/compiler_types.h -D__KERNEL__ -fmacro-prefix-map=/nix/store/w2bzm4kvp46xhsqpbqvp1yyxmp5zq0fz-linux-6.6.30-dev/lib/modules/6.6.30/source/= -std=gnu11 -fshort-wchar -funsigned-char -fno-common -fno-PIE -fno-strict-aliasing -mno-sse -mno-mmx -mno-sse2 -mno-3dnow -mno-avx -fcf-protection=branch -fno-jump-tables -m64 -falign-jumps=1 -falign-loops=1 -mno-80387 -mno-fp-ret-in-387 -mpreferred-stack-boundary=3 -mskip-rax-setup -mtune=generic -mno-red-zone -mcmodel=kernel -fno-asynchronous-unwind-tables -mindirect-branch=thunk-extern -mindirect-branch-register -mindirect-branch-cs-prefix -mfunction-return=thunk-extern -fno-jump-tables -mharden-sls=all -fpatchable-function-entry=16,16 -fno-delete-null-pointer-checks -O2 -fno-allow-store-data-races -fstack-protector-strong -ftrivial-auto-var-init=zero -fno-stack-clash-protection -pg -mrecord-mcount -mfentry -DCC_USING_FENTRY -falign-functions=16 -fstrict-flex-arrays=3 -fno-strict-overflow -fno-stack-check -fconserve-stack -g -DMODULE -DKBUILD_BASENAME="main" -DKBUILD_MODNAME="main" -D__KBUILD_MODNAME=kmod_main -fno-pie -Werror=incompatible-pointer-types -Wno-nonnull -Werror=implicit -isystem /nix/store/9hgsinpfgyvsd92v0wlvmxv9wnaal68r-gcc-13.2.0/lib/gcc/x86_64-unknown-linux-gnu/13.2.0/include
ena> ### Detection of the available kernel API ###
ena> ENA_HAVE_PCI_DEV_ID: YES
ena> ENA_HAVE_XDP_DO_FLUSH: YES
ena> ENA_HAVE_CPUMASK_LOCAL_SPREAD: YES
[... 23 lines cut ...]
ena> ERROR: modpost: "ethtool_puts" [/build/source/kernel/linux/ena/ena.ko] undefined!
ena> make[3]: *** [/nix/store/w2bzm4kvp46xhsqpbqvp1yyxmp5zq0fz-linux-6.6.30-dev/lib/modules/6.6.30/source/scripts/Makefile.modpost:145: /build/source/kernel/linux/ena/Module.symvers] Error 1
ena> make[2]: *** [/nix/store/w2bzm4kvp46xhsqpbqvp1yyxmp5zq0fz-linux-6.6.30-dev/lib/modules/6.6.30/source/Makefile:1865: modpost] Error 2
ena> make[1]: *** [/nix/store/w2bzm4kvp46xhsqpbqvp1yyxmp5zq0fz-linux-6.6.30-dev/lib/modules/6.6.30/source/Makefile:234: __sub-make] Error 2
ena> make[1]: Leaving directory '/nix/store/w2bzm4kvp46xhsqpbqvp1yyxmp5zq0fz-linux-6.6.30-dev/lib/modules/6.6.30/build'
ena> make: *** [Makefile:102: ena.ko] Error 2
error: builder for '/nix/store/blxa9qml79w7lrm0h6d2vw07l57w39l9-ena-2.12.0-6.6.30.drv' failed with exit code 2;
       last 10 log lines:
       >   CC [M]  /build/source/kernel/linux/ena/../common/ena_com//ena_eth_com.o
       >   CC [M]  /build/source/kernel/linux/ena/ena_sysfs.o
       >   LD [M]  /build/source/kernel/linux/ena/ena.o
       >   MODPOST /build/source/kernel/linux/ena/Module.symvers
       > ERROR: modpost: "ethtool_puts" [/build/source/kernel/linux/ena/ena.ko] undefined!
       > make[3]: *** [/nix/store/w2bzm4kvp46xhsqpbqvp1yyxmp5zq0fz-linux-6.6.30-dev/lib/modules/6.6.30/source/scripts/Makefile.modpost:145: /build/source/kernel/linux/ena/Module.symvers] Error 1
       > make[2]: *** [/nix/store/w2bzm4kvp46xhsqpbqvp1yyxmp5zq0fz-linux-6.6.30-dev/lib/modules/6.6.30/source/Makefile:1865: modpost] Error 2
       > make[1]: *** [/nix/store/w2bzm4kvp46xhsqpbqvp1yyxmp5zq0fz-linux-6.6.30-dev/lib/modules/6.6.30/source/Makefile:234: __sub-make] Error 2
       > make[1]: Leaving directory '/nix/store/w2bzm4kvp46xhsqpbqvp1yyxmp5zq0fz-linux-6.6.30-dev/lib/modules/6.6.30/build'
       > make: *** [Makefile:102: ena.ko] Error 2
       For full logs, run 'nix log /nix/store/blxa9qml79w7lrm0h6d2vw07l57w39l9-ena-2.12.0-6.6.30.drv'.
</code>
</details>

### Contact Details

me@enric.me

davidarinzon, 2024-05-15:
Thank you for raising this @kiike 
We will look into this.

If you have relevant information such as AMI ID and additional steps that might help us, please share them here.

ShayAgros, 2024-05-16:
Hi,
I'm trying to reproduce the issue. Note that _I'm not using NixOS_.

I did the following:
Downloaded vanilla kernel version 6.6.30 and compiled it:
```
$ wget https://cdn.kernel.org/pub/linux/kernel/v6.x/linux-6.6.30.tar.xz 
$ cd linux-6.6.30
$ cp /boot/config-`uname -r` ./.config
$ make olddefconfig
$ ./scripts/config -e CONFIG_NET_VENDOR_AMAZON
$ ./scripts/config -m CONFIG_ENA_ETHERNET
$ make -j`nproc`
```

Downloading the Github driver and compiled it against this kernel:
```
$ git clone https://github.com/amzn/amzn-drivers.git ~/amzn-drivers
$ cd ~/amzn-drivers/kernel/linux/ena
$ KERNEL_BUILD_DIR=~/linux-6.6.30 make
```

The compilation does seem to succeed:
```
$ KERNEL_BUILD_DIR=~/linux-6.6.30 make
./configure.sh --kernel-dir /home/ec2-user/linux-6.6.30 --verbose
Trying to intercept CC arguments
Detected compiler:  gcc
Managed to intercept CC arguments
Command: gcc -nostdinc -I./arch/x86/include -I./arch/x86/include/generated -I./include -I./arch/x86/include/uapi -I./arch/x86/include/generated/uapi -I./include/uapi -I./include/generated/uapi -include ./include/linux/compiler-version.h -include ./include/linux/kconfig.h -include ./include/linux/compiler_types.h -D__KERNEL__ -fmacro-prefix-map=./= -std=gnu11 -fshort-wchar -funsigned-char -fno-common -fno-PIE -fno-strict-aliasing -mno-sse -mno-mmx -mno-sse2 -mno-3dnow -mno-avx -fcf-protection=none -m64 -falign-jumps=1 -falign-loops=1 -mno-80387 -mno-fp-ret-in-387 -mpreferred-stack-boundary=3 -mskip-rax-setup -mtune=generic -mno-red-zone -mcmodel=kernel -fno-asynchronous-unwind-tables -mindirect-branch=thunk-extern -mindirect-branch-register -mindirect-branch-cs-prefix -mfunction-return=thunk-extern -fno-jump-tables -fpatchable-function-entry=16,16 -fno-delete-null-pointer-checks -O2 -fno-allow-store-data-races -fstack-protector-strong -fno-stack-clash-protection -pg -mrecord-mcount -mfentry -DCC_USING_FENTRY -fno-inline-functions-called-once -falign-functions=16 -fno-strict-overflow -fno-stack-check -fconserve-stack -g -gdwarf-4 -DMODULE -DKBUILD_BASENAME="main" -DKBUILD_MODNAME="main" -D__KBUILD_MODNAME=kmod_main -fno-pie -Werror=incompatible-pointer-types -Wno-nonnull -Werror=implicit
### Detection of the available kernel API ###
ENA_HAVE_PCI_DEV_ID: YES
ENA_HAVE_XDP_DO_FLUSH: YES
ENA_HAVE_CPUMASK_LOCAL_SPREAD: YES
ENA_HAVE_UPDATE_AFFINITY_HINT: YES
ENA_HAVE_ETHTOOL_PUTS: NO
ENA_HAVE_ETHTOOL_RXFH_PARAM: NO
### Detection completed ###
make -C /home/ec2-user/linux-6.6.30 M=/home/ec2-user/amzn-drivers/kernel/linux/ena modules
make[1]: Entering directory '/home/ec2-user/linux-6.6.30'
  CC [M]  /home/ec2-user/amzn-drivers/kernel/linux/ena/ena_netdev.o
  CC [M]  /home/ec2-user/amzn-drivers/kernel/linux/ena/ena_ethtool.o
  CC [M]  /home/ec2-user/amzn-drivers/kernel/linux/ena/ena_lpc.o
  CC [M]  /home/ec2-user/amzn-drivers/kernel/linux/ena/ena_phc.o
  CC [M]  /home/ec2-user/amzn-drivers/kernel/linux/ena/ena_xdp.o
  CC [M]  /home/ec2-user/amzn-drivers/kernel/linux/ena/dim.o
  CC [M]  /home/ec2-user/amzn-drivers/kernel/linux/ena/net_dim.o
  CC [M]  /home/ec2-user/amzn-drivers/kernel/linux/ena/../common/ena_com//ena_com.o
  CC [M]  /home/ec2-user/amzn-drivers/kernel/linux/ena/../common/ena_com//ena_eth_com.o
  CC [M]  /home/ec2-user/amzn-drivers/kernel/linux/ena/ena_sysfs.o
  LD [M]  /home/ec2-user/amzn-drivers/kernel/linux/ena/ena.o
  MODPOST /home/ec2-user/amzn-drivers/kernel/linux/ena/Module.symvers
  CC [M]  /home/ec2-user/amzn-drivers/kernel/linux/ena/ena.mod.o
  LD [M]  /home/ec2-user/amzn-drivers/kernel/linux/ena/ena.ko
  BTF [M] /home/ec2-user/amzn-drivers/kernel/linux/ena/ena.ko
make[1]: Leaving directory '/home/ec2-user/linux-6.6.30'
```

It is possible that we're using a different version of Linux 6.6.30 or a compiler variance is what causing the check of `ethtool_puts` to falsely succeed, so I'd appreciate if you can provide more information for them

kiike, 2024-05-16:
Thanks for trying to reproduce it. The build log contains the intercepted compiler command calls as well as the versions of everything used to try the build. I noticed that there might be something erroneous when calling the build, because the feature detection always reports the same available features, no matter what kernel version I am pointing it to. For instance, for a 4.19 kernel, which should have less available features, I am getting:

ena> ### Detection of the available kernel API ###
ena> ENA_HAVE_PCI_DEV_ID: YES
ena> ENA_HAVE_XDP_DO_FLUSH: YES
ena> ENA_HAVE_CPUMASK_LOCAL_SPREAD: YES
ena> ENA_HAVE_UPDATE_AFFINITY_HINT: YES
ena> ENA_HAVE_ETHTOOL_PUTS: YES
ena> ENA_HAVE_ETHTOOL_RXFH_PARAM: NO
ena> ### Detection completed ###

If I am running the same long command on the git repo, the reported features are more in line what would be expected, although the build fails.

```
### Detection of the available kernel API ###
ENA_HAVE_PCI_DEV_ID: NO
ENA_HAVE_XDP_DO_FLUSH: NO
ENA_HAVE_CPUMASK_LOCAL_SPREAD: YES
ENA_HAVE_UPDATE_AFFINITY_HINT: NO
ENA_HAVE_ETHTOOL_PUTS: NO
ENA_HAVE_ETHTOOL_RXFH_PARAM: NO
### Detection completed ###
make -C /nix/store/sfd9v1slhdwcz64hn4rl8802y2b7i7fh-linux-4.19.313-dev/lib/modules/4.19.313/build M=/home/kiike/projects/amzn-drivers/kernel/linux/ena modules
  CC [M]  /home/kiike/projects/amzn-drivers/kernel/linux/ena/ena_netdev.o
In file included from /home/kiike/projects/amzn-drivers/kernel/linux/ena/ena_netdev.h:9,
                 from /home/kiike/projects/amzn-drivers/kernel/linux/ena/ena_netdev.c:25:
/home/kiike/projects/amzn-drivers/kernel/linux/ena/kcompat.h:900:20: error: redefinition of ‘eth_hw_addr_set’
  900 | static inline void eth_hw_addr_set(struct net_device *dev, const u8 *addr)
```
As a workaround, I am using patches on the kcompat.h file to use the kernel version as a hint.

ShayAgros, 2024-05-19:
> The build log contains the intercepted compiler command calls as well as the versions of everything used to try the build

Yup it is, just picking up a NixOS distribution randomly doesn't seem to give the the same environment as yours. It'd be quicker if you send me the ami-id (+region) and the nix configuration you're using.

> If I am running the same long command on the git repo, the reported features are more in line what would be expected, although the build fails.

Can you please send me the commands you're using ? Did you try running the commands I used ?

> because the feature detection always reports the same available features, no matter what kernel version I am pointing it to.

I'm not sure whether you're using the build instructions for the nix package I found:
https://github.com/NixOS/nixpkgs/blob/nixos-23.11/pkgs/os-specific/linux/ena/default.nix#L26

but if so, note that replacing the kernel directory in the Makefile itself is not the intended way to modify the kernel version.

To build the driver against a specific kernel directory, you need to set the environment variable `KERNEL_BUILD_DIR` like I did in my previous comment:
```
$ export KERNEL_BUILD_DIR=${HOME}/linux-6.6.30
$ make
```

Maybe using this approach would solve the compilation error

kiike, 2024-05-20:
AMI: ami-07158550d1e5327f0
AMI name: nixos/23.11.7071.71bae31b7dbc-x86_64-linux
Region: eu-west-3

I am using a build derivation similar to what you linked. It's on [nixpkgs master](https://github.com/NixOS/nixpkgs/blob/master/pkgs/os-specific/linux/ena/default.nix). To test building, I am now removing the api features detection patch I included there, as well as an exported env variable with the build dir as per your recommendation.

<details>
<summary>Here are the contents of the derivation at <code>pkgs/os-specific/linux/ena/default.nix</code></summary>
<code>{ lib, stdenv, fetchFromGitHub, kernel }:
stdenv.mkDerivation rec {
  version = "2.12.0";
  name = "ena-${version}-${kernel.version}";
  src = fetchFromGitHub {
    owner = "amzn";
    repo = "amzn-drivers";
    rev = "ena_linux_${version}";
    hash = "sha256-Z/eeIUY7Yl2l/IqK3Z2nxPhn+JLvP976IZ9ZXPBqoSo=";
  };
  hardeningDisable = [ "pic" ];
  nativeBuildInputs = kernel.moduleBuildDependencies;
  makeFlags = kernel.makeFlags;
  # linux 3.12
  env.NIX_CFLAGS_COMPILE = "-Wno-error=implicit-function-declaration";
  patches = [
    # Use kernel version checks instead of API feature detection
    # See https://github.com/NixOS/nixpkgs/pull/310680
    # ./override-features-api-detection.patch
  ];
  configurePhase = ''
    runHook preConfigure
    cd kernel/linux/ena
    export ENA_PHC_INCLUDE=1 KERNEL_BUILD_DIR=${kernel.dev}/lib/modules/${kernel.modDirVersion}/build
    runHook postConfigure
  '';
  installPhase = ''
    runHook preInstall
    $STRIP -S ena.ko
    dest=$out/lib/modules/${kernel.modDirVersion}/misc
    mkdir -p $dest
    cp ena.ko $dest/
    xz $dest/ena.ko
    runHook postInstall
  '';
  meta = with lib; {
    description = "Amazon Elastic Network Adapter (ENA) driver for Linux";
    homepage = "https://github.com/amzn/amzn-drivers";
    license = licenses.gpl2Only;
    maintainers = with maintainers; [ eelco sielicki ];
    platforms = platforms.linux;
  };
}</code>
</details>

To trigger the build, I am using the command `nix-build -A linuxPackages.ena -v --keep-failed` inside the cloned `nixpkgs` repository on the `master` branch. This builds for the current default kernel which is 6.6.31, but it's possible to also build the module for 6.8 with `nix-build -A linuxPackages_6_8.ena`, or  `nix-build -A linuxPackages_4_19.ena` for kernel 4.19. Unfortunately, exporting the `KERNEL_BUILD_DIR` environment variable doesn't result in the build succeeding.

Thank you for your support on this. Please feel free to close the issue if you think it deviates from the supported path.

ShayAgros, 2024-05-22:
Ok, I found the culprit, which is the line
```
env.NIX_CFLAGS_COMPILE = "-Wno-error=implicit-function-declaration";
```

First, a short explanation about the build system which is called ECC (ENA Compatibility Checker):
The system tries to identify the existence of a specific code by trying to compile a small snippet. The assumption here is that the snippet compiles iff the feature is supported by the kernel (we run regular checks to verify that this is indeed the case).

Back to the problem in this ticket:
When ECC checks for the support of `ethtool_puts` function it tries to run the following code snippet:
```
#include <linux/module.h>
#include <linux/kernel.h>
#include <linux/version.h>
#define ENA_REQUIRE_MIN_VERSION(A, B, C) LINUX_VERSION_CODE < KERNEL_VERSION(A, B, C)
MODULE_LICENSE("Dual BSD/GPL");
#include <linux/ethtool.h>
int
main(void)
{ethtool_puts(NULL, NULL);
;
return 0;
}
```

Which for NixOS case returned the compilation output
```
/build/amzn-drivers/kernel/linux/ena/tmp_ENA_HAVE_ETHTOOL_PUTS_1716303999638_/main.c: In function 'main':
/build/amzn-drivers/kernel/linux/ena/tmp_ENA_HAVE_ETHTOOL_PUTS_1716303999638_/main.c:9:2: warning: implicit declaration of function 'ethtool_puts'; did you mean 'ethtool_sprintf'? [-Wimplicit-function-declaration]
    9 | {ethtool_puts(NULL, NULL);
      |  ^~~~~~~~~~~~
      |  ethtool_sprintf
```

This output should have returned an error (as ECC runs the compiler with `-Werror=implicit`), however the configuration file prevents failing this test because of this warning.

Removing the line
```
env.NIX_CFLAGS_COMPILE = "-Wno-error=implicit-function-declaration";
```

makes the compilation succeed
```
./configure.sh --kernel-dir /nix/store/akphn3a1a37jcnz317pph0kynzm1c7v3-linux-6.6.31-dev/lib/modules/6.6.31/build --verbose
Trying to intercept CC arguments
Detected compiler:  /nix/store/md6hh4rrcrf99nssvcam3qaqs3skj086-gcc-wrapper-13.2.0/bin/cc
Managed to intercept CC arguments
Command: /nix/store/md6hh4rrcrf99nssvcam3qaqs3skj086-gcc-wrapper-13.2.0/bin/cc -nostdinc -I/nix/store/akphn3a1a37jcnz317pph0kynzm1c7v3-linux-6.6.31-dev/lib/modules/6.6.31/source/arch/x86/include -I./arch/x86/include/generated -I/nix/store/akphn3a1a37jcnz317pph0kynzm1c7v3-linux-6.6.31-dev/lib/modules/6.6.31/source/include -I./include -I/nix/store/akphn3a1a37jcnz317pph0kynzm1c7v3-linux-6.6.31-dev/lib/modules/6.6.31/source/arch/x86/include/uapi -I./arch/x86/include/generated/uapi -I/nix/store/akphn3a1a37jcnz317pph0kynzm1c7v3-linux-6.6.31-dev/lib/modules/6.6.31/source/include/uapi -I./include/generated/uapi -include /nix/store/akphn3a1a37jcnz317pph0kynzm1c7v3-linux-6.6.31-dev/lib/modules/6.6.31/source/include/linux/compiler-version.h -include /nix/store/akphn3a1a37jcnz317pph0kynzm1c7v3-linux-6.6.31-dev/lib/modules/6.6.31/source/include/linux/kconfig.h -include /nix/store/akphn3a1a37jcnz317pph0kynzm1c7v3-linux-6.6.31-dev/lib/modules/6.6.31/source/include/linux/compiler_types.h -D__KERNEL__ -fmacro-prefix-map=/nix/store/akphn3a1a37jcnz317pph0kynzm1c7v3-linux-6.6.31-dev/lib/modules/6.6.31/source/= -std=gnu11 -fshort-wchar -funsigned-char -fno-common -fno-PIE -fno-strict-aliasing -mno-sse -mno-mmx -mno-sse2 -mno-3dnow -mno-avx -fcf-protection=branch -fno-jump-tables -m64 -falign-jumps=1 -falign-loops=1 -mno-80387 -mno-fp-ret-in-387 -mpreferred-stack-boundary=3 -mskip-rax-setup -mtune=generic -mno-red-zone -mcmodel=kernel -fno-asynchronous-unwind-tables -mindirect-branch=thunk-extern -mindirect-branch-register -mindirect-branch-cs-prefix -mfunction-return=thunk-extern -fno-jump-tables -mharden-sls=all -fpatchable-function-entry=16,16 -fno-delete-null-pointer-checks -O2 -fno-allow-store-data-races -fstack-protector-strong -ftrivial-auto-var-init=zero -fno-stack-clash-protection -pg -mrecord-mcount -mfentry -DCC_USING_FENTRY -falign-functions=16 -fstrict-flex-arrays=3 -fno-strict-overflow -fno-stack-check -fconserve-stack -g -DMODULE -DKBUILD_BASENAME="main" -DKBUILD_MODNAME="main" -D__KBUILD_MODNAME=kmod_main -fno-pie -Werror=incompatible-pointer-types -Wno-nonnull -Werror=implicit -isystem /nix/store/9hgsinpfgyvsd92v0wlvmxv9wnaal68r-gcc-13.2.0/lib/gcc/x86_64-unknown-linux-gnu/13.2.0/include
### Detection of the available kernel API ###
ENA_HAVE_PCI_DEV_ID: YES
ENA_HAVE_XDP_DO_FLUSH: YES
ENA_HAVE_CPUMASK_LOCAL_SPREAD: YES
ENA_HAVE_UPDATE_AFFINITY_HINT: YES
ENA_HAVE_ETHTOOL_PUTS: NO
ENA_HAVE_ETHTOOL_RXFH_PARAM: NO
### Detection completed ###
make -C /nix/store/akphn3a1a37jcnz317pph0kynzm1c7v3-linux-6.6.31-dev/lib/modules/6.6.31/build M=/build/source/kernel/linux/ena modules
make[1]: Entering directory '/nix/store/akphn3a1a37jcnz317pph0kynzm1c7v3-linux-6.6.31-dev/lib/modules/6.6.31/build'
  CC [M]  /build/source/kernel/linux/ena/ena_netdev.o
  CC [M]  /build/source/kernel/linux/ena/ena_ethtool.o
  CC [M]  /build/source/kernel/linux/ena/ena_lpc.o
  CC [M]  /build/source/kernel/linux/ena/ena_phc.o
  CC [M]  /build/source/kernel/linux/ena/ena_xdp.o
  CC [M]  /build/source/kernel/linux/ena/dim.o
  CC [M]  /build/source/kernel/linux/ena/net_dim.o
  CC [M]  /build/source/kernel/linux/ena/../common/ena_com//ena_com.o
  CC [M]  /build/source/kernel/linux/ena/../common/ena_com//ena_eth_com.o
  CC [M]  /build/source/kernel/linux/ena/ena_sysfs.o
  LD [M]  /build/source/kernel/linux/ena/ena.o
  MODPOST /build/source/kernel/linux/ena/Module.symvers
  CC [M]  /build/source/kernel/linux/ena/ena.mod.o
  LD [M]  /build/source/kernel/linux/ena/ena.ko
  BTF [M] /build/source/kernel/linux/ena/ena.ko
[... cut ...]
patching script interpreter paths in /nix/store/vpnggaamm20k9gg6h4zabrspp69v8psf-ena-2.12.0-6.6.31
stripping (with command strip and flags -S -p) in  /nix/store/vpnggaamm20k9gg6h4zabrspp69v8psf-ena-2.12.0-6.6.31/lib
/nix/store/vpnggaamm20k9gg6h4zabrspp69v8psf-ena-2.12.0-6.6.31
```

Going further from here:
I don't know the reason for disabling this warning in the package, but please note that this breaks our ECC system. We don't recommended tweaking the compiler error handling when compiling ENA driver

kiike, 2024-05-22:
Thank you so much for researching this further! Your explanation makes sense and I now understand better the reasoning for the error. Indeed, disabling this line enables successful detection of the features and building for the following packaged kernels for NixOS:

- 6.9.1
- 6.8.9
- 6.6.31
- 5.15.159
- 5.4.276

Only 4.19.314 is failing:

```
ena> ### Detection of the available kernel API ###
ena> ENA_HAVE_PCI_DEV_ID: NO
ena> ENA_HAVE_XDP_DO_FLUSH: NO
ena> ENA_HAVE_CPUMASK_LOCAL_SPREAD: YES
ena> ENA_HAVE_UPDATE_AFFINITY_HINT: NO
ena> ENA_HAVE_ETHTOOL_PUTS: NO
ena> ENA_HAVE_ETHTOOL_RXFH_PARAM: NO
ena> ### Detection completed ###
ena> make -C /nix/store/rdhpl75hqbdpqw52d3gs210i6zs2wwlq-linux-4.19.314-dev/lib/modules/4.19.314/build M=/build/source/kernel/linux/ena modules
ena>   CC [M]  /build/source/kernel/linux/ena/ena_netdev.o
ena> In file included from /build/source/kernel/linux/ena/ena_netdev.h:9,
ena>                  from /build/source/kernel/linux/ena/ena_netdev.c:25:
ena> /build/source/kernel/linux/ena/kcompat.h:900:20: error: redefinition of 'eth_hw_addr_set'
ena>   900 | static inline void eth_hw_addr_set(struct net_device *dev, const u8 *addr)
ena>       |                    ^~~~~~~~~~~~~~~
ena> compilation terminated due to -Wfatal-errors.
ena> make[4]: *** [/nix/store/rdhpl75hqbdpqw52d3gs210i6zs2wwlq-linux-4.19.314-dev/lib/modules/4.19.314/source/scripts/Makefile.build:304: /build/source/kernel/linux/ena/ena_netdev.o] Error 1
ena> make[3]: *** [/nix/store/rdhpl75hqbdpqw52d3gs210i6zs2wwlq-linux-4.19.314-dev/lib/modules/4.19.314/source/Makefile:1562: _module_/build/source/kernel/linux/ena] Error 2
ena> make[2]: *** [Makefile:153: sub-make] Error 2
ena> make[1]: *** [Makefile:24: __sub-make] Error 2
ena> make: *** [Makefile:102: ena.ko] Error 2
error: builder for '/nix/store/7ilajxfn7pl9dy9zk4w6ig0vbhr0fwv9-ena-2.12.0-4.19.314.drv' failed with exit code 2
```
Regarding that flag, I think it can safely go, as 3.12 is not available anymore for NixOS.

kiike, 2024-05-23:
The patch works perfectly and all other kernel versions compile successfully as you indicated. I am trying to push for using it in https://github.com/NixOS/nixpkgs/pull/313941. Thanks again!

closed 2024-05-23 by kiike