amzn/amzn-drivers #303: [Bug]: Compilation of ena_linux v2.12.0 for kernels > 6.8.4 (even 6.9.1)

aledbf, 2024-05-18:
### Preliminary Actions

- [X] I have searched the existing [issues](https://github.com/amzn/amzn-drivers/issues) and didn't find a duplicate.
- [X] I have followed the [AWS official troubleshoot documentation](https://docs.aws.amazon.com/AWSEC2/latest/UserGuide/troubleshooting-ena.html).
- [X] I have followed the driver [readme](https://github.com/amzn/amzn-drivers) and [best practices](https://github.com/amzn/amzn-drivers/blob/master/kernel/linux/ena/ENA_Linux_Best_Practices.rst).

### Driver Type

Linux kernel driver for Elastic Network Adapter (ENA)

### Driver Tag/Commit

ena_linux_2.12.0

### Custom Code

No

### OS Platform and Distribution

Linux ip-172-31-2-238 6.8.9 #1 SMP PREEMPT_DYNAMIC Sat May 18 18:39:27 UTC 2024 x86_64 x86_64 x86_64 GNU/Linux

### Bug description

```
./configure.sh 
Trying to intercept CC arguments
Detected compiler:  gcc
Managed to intercept CC arguments
### Detection of the available kernel API ###
### Detection completed ###
root@ip-172-31-2-238:/usr/src/amzn-drivers-1.0.0/kernel/linux/ena# make
make -C /lib/modules/6.8.9/build M=/usr/src/amzn-drivers-1.0.0/kernel/linux/ena modules
make[1]: Entering directory '/usr/src/linux-headers-6.8.9'
  CC [M]  /usr/src/amzn-drivers-1.0.0/kernel/linux/ena/ena_netdev.o
/usr/src/amzn-drivers-1.0.0/kernel/linux/ena/ena_netdev.c:4040:34: error: no previous prototype for ‘check_cdesc_in_tx_cq’ [-Werror=missing-prototypes]
 4040 | enum ena_regs_reset_reason_types check_cdesc_in_tx_cq(struct ena_adapter *adapter,
      |                                  ^~~~~~~~~~~~~~~~~~~~
compilation terminated due to -Wfatal-errors.
cc1: all warnings being treated as errors
make[3]: *** [scripts/Makefile.build:243: /usr/src/amzn-drivers-1.0.0/kernel/linux/ena/ena_netdev.o] Error 1
make[2]: *** [/usr/src/linux-headers-6.8.9/Makefile:1927: /usr/src/amzn-drivers-1.0.0/kernel/linux/ena] Error 2
make[1]: *** [Makefile:240: __sub-make] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-6.8.9'
make: *** [Makefile:102: ena.ko] Error 2
```

### Reproduction steps

```bash
cd kernel/linux/ena/
./configure.sh
```

### Expected Behavior

working kernel module

### Actual Behavior

build error

davidarinzon, 2024-05-19:
Thank you for raising this issue, @aledbf 
We will look into it.

davidarinzon, 2024-05-19:
@aledbf 
We will release a fix for this issue soon, if you'd like to receive a W/A patch in the meanwhile, please let us know.

aledbf, 2024-05-25:
@davidarinzon It's working as expected. Thank you for the quick fix!

closed 2024-05-25 by aledbf