amzn/amzn-drivers #305: [Support]: coalescing features

likewise, 2024-06-18:
### Preliminary Actions

- [X] I have searched the existing [issues](https://github.com/amzn/amzn-drivers/issues) and didn't find a duplicate.
- [X] I have followed the [AWS official troubleshoot documentation](https://docs.aws.amazon.com/AWSEC2/latest/UserGuide/troubleshooting-ena.html).
- [X] I have followed the driver [readme](https://github.com/amzn/amzn-drivers) and [best practices](https://github.com/amzn/amzn-drivers/blob/master/kernel/linux/ena/ENA_Linux_Best_Practices.rst).

### Driver Type

Linux kernel driver for Elastic Network Adapter (ENA)

### Driver Tag/Commit

ena_linux_2.11.1

### Custom Code

No

### OS Platform and Distribution

different Linux kernel versions, custom and default

### Support request

What interrupt coalescing options should be available with the latest ENA?  
Was "adaptive TX" ever supported for the Linux kernel driver?

We have a 5.10 kernel that shows all the options available, and are confused, because the (current) driver source code does not seem to support these. Or are these implemented in-kernel rather than in-driver??

Coalesce parameters for ens7:
Adaptive RX: off  TX: off
stats-block-usecs: 0
sample-interval: 0
pkt-rate-low: 0
pkt-rate-high: 0

rx-usecs: 0
rx-frames: 0
rx-usecs-irq: 0
rx-frames-irq: 0

tx-usecs: 0
tx-frames: 0
tx-usecs-irq: 0
tx-frames-irq: 0

rx-usecs-low: 0
rx-frame-low: 0
tx-usecs-low: 0
tx-frame-low: 0

rx-usecs-high: 0
rx-frame-high: 0
tx-usecs-high: 0
tx-frame-high: 0

### Contact Details

leon@sidebranch.com

davidarinzon, 2024-06-18:
Hi @likewise 

Adaptive TX moderation was never supported, only Adaptive RX moderation is supported and is enabled by default in the driver.
Though execution of ethtool may show all of these parameters, you should focus on `rx-usecs`, `tx-usecs` and `Adaptive RX`.
ethtool outputs all the standard information it has, and each driver fills the fields that it supports, though ethtool still outputs all of the fields.

I do see that all of the values are 0 in your case, I assume you've loaded the github driver and have tried manipulating some of the values (Like disabling interrupt moderation, as suggested in our best practices for some use-cases)?

Would you like to share your use-case and what you're trying to achieve?

likewise, 2024-06-24:
Use case is quick response time to RX for low traffic, either quick interrupts or no interrupts and busy polling I guess.

davidarinzon, 2024-06-25:
Hi @likewise 
Please check the best practices page, there's a question about best latency there (`Q: What are the optimal settings for achieving the best latency`), I believe that the steps there will be helpful.
Let us know otherwise.

likewise, 2024-07-25:
Thanks for explaining what was/is implemented vs. what is shown.

closed 2024-07-25 by likewise