amzn/amzn-drivers #313: [Bug]: Does not build on Kernel 6.10

arianvp, 2024-07-27:
### Preliminary Actions

- [X] I have searched the existing [issues](https://github.com/amzn/amzn-drivers/issues) and didn't find a duplicate.
- [X] I have followed the [AWS official troubleshoot documentation](https://docs.aws.amazon.com/AWSEC2/latest/UserGuide/troubleshooting-ena.html).
- [X] I have followed the driver [readme](https://github.com/amzn/amzn-drivers) and [best practices](https://github.com/amzn/amzn-drivers/blob/master/kernel/linux/ena/ENA_Linux_Best_Practices.rst).

### Driver Type

Linux kernel driver for Elastic Network Adapter (ENA)

### Driver Tag/Commit

ena_linux_2.12.3

### Custom Code

No

### OS Platform and Distribution

Linux 6.10.1

### Bug description

The driver does not compile on Kernel 6.10.

This seems to be caused by the renaming of `__napi_alloc_skb` to `napi_alloc_skb` https://github.com/torvalds/linux/commit/6e9b01909a811555ff3326cf80a5847169c57806

### Reproduction steps

```bash
Build the driver with the 6.10 kernel
```

### Expected Behavior

it compiles

### Actual Behavior

It doesn't compile

### Relevant log output

```shell
make -C /nix/store/2yg16mg5cgcckxl0hjzvhz7sw3yrkwjm-linux-6.10.1-dev/lib/modules/6.10.1/build M=/build/source/kernel/linux/ena modules
make[1]: Entering directory '/nix/store/2yg16mg5cgcckxl0hjzvhz7sw3yrkwjm-linux-6.10.1-dev/lib/modules/6.10.1/build'
  CC [M]  /build/source/kernel/linux/ena/ena_netdev.o
  CC [M]  /build/source/kernel/linux/ena/ena_ethtool.o
  CC [M]  /build/source/kernel/linux/ena/ena_lpc.o
  CC [M]  /build/source/kernel/linux/ena/ena_phc.o
  CC [M]  /build/source/kernel/linux/ena/ena_xdp.o
/build/source/kernel/linux/ena/ena_xdp.c: In function 'ena_xdp_rx_skb_zc':
/build/source/kernel/linux/ena/ena_xdp.c:749:15: error: implicit declaration of function '__napi_alloc_skb'; did you mean 'napi_alloc_skb'? [-Werror=implicit-function-declaration]
  749 |         skb = __napi_alloc_skb(rx_ring->napi,
      |               ^~~~~~~~~~~~~~~~
      |               napi_alloc_skb
compilation terminated due to -Wfatal-errors.
cc1: some warnings being treated as errors
```

davidarinzon, 2024-07-28:
Hi @arianvp 
Thank you for raising this issue, we are aware of it, and it will be addressed in an upcoming release of the driver.
We will provide a slightly more generic solution that will resolve this issue for future OS distributions.

As you have proposed a patch, I assume that the issue is not blocking you from continuing your AF_XDP development.

arianvp, 2024-07-28:
Nope all good. The patch unblocks me for now. Thanks

arianvp, 2024-07-29:
Ah after my patch I run into the next compiler error:

```
error: builder for '/nix/store/lxmbs6ya60x68wf5a8mpk5njg3f0xrwh-ena-2.12.3-6.10.2.drv' failed with exit code 2;
       last 10 log lines:
       > /build/source/kernel/linux/ena/ena_xdp.c: In function 'ena_xdp_clean_rx_irq_zc':
       > /build/source/kernel/linux/ena/ena_xdp.c:816:17: error: too many arguments to function 'xsk_buff_dma_sync_for_cpu'
       >   816 |                 xsk_buff_dma_sync_for_cpu(xdp, rx_ring->xsk_pool);
       >       |                 ^~~~~~~~~~~~~~~~~~~~~~~~~
       > compilation terminated due to -Wfatal-errors.
       > make[3]: *** [/nix/store/216k8yzq05s5r3lvnh7jd9hbric4cmln-linux-libre-6.10.2-dev/lib/modules/6.10.2-gnu/source/scripts/Makefile.build:244: /build/source/kernel/linux/ena/ena_xdp.o] Error 1
       > make[2]: *** [/nix/store/216k8yzq05s5r3lvnh7jd9hbric4cmln-linux-libre-6.10.2-dev/lib/modules/6.10.2-gnu/source/Makefile:1934: /build/source/kernel/linux/ena] Error 2
       > make[1]: *** [/nix/store/216k8yzq05s5r3lvnh7jd9hbric4cmln-linux-libre-6.10.2-dev/lib/modules/6.10.2-gnu/source/Makefile:240: __sub-make] Error 2
       > make[1]: Leaving directory '/nix/store/216k8yzq05s5r3lvnh7jd9hbric4cmln-linux-libre-6.10.2-dev/lib/modules/6.10.2-gnu/build'
       > make: *** [Makefile:102: ena.ko] Error 2
       For full logs, run 'nix log /nix/store/lxmbs6ya60x68wf5a8mpk5njg3f0xrwh-ena-2.12.3-6.10.2.drv'.
error: builder for '/nix/store/j23sjz0wq8gq3y9p8a6082r0nvg9qnyi-ena-2.12.3-6.10.2.drv' failed with exit code 2;
       last 10 log lines:
       > /build/source/kernel/linux/ena/ena_xdp.c: In function 'ena_xdp_clean_rx_irq_zc':
       > /build/source/kernel/linux/ena/ena_xdp.c:816:17: error: too many arguments to function 'xsk_buff_dma_sync_for_cpu'
       >   816 |                 xsk_buff_dma_sync_for_cpu(xdp, rx_ring->xsk_pool);
       >       |                 ^~~~~~~~~~~~~~~~~~~~~~~~~
       > compilation terminated due to -Wfatal-errors.
       > make[3]: *** [/nix/store/4pi3spm0m1w4hyr659wkpxmf9cj7b6zr-linux-6.10.2-dev/lib/modules/6.10.2/source/scripts/Makefile.build:244: /build/source/kernel/linux/ena/ena_xdp.o] Error 1
       > make[2]: *** [/nix/store/4pi3spm0m1w4hyr659wkpxmf9cj7b6zr-linux-6.10.2-dev/lib/modules/6.10.2/source/Makefile:1934: /build/source/kernel/linux/ena] Error 2
       > make[1]: *** [/nix/store/4pi3spm0m1w4hyr659wkpxmf9cj7b6zr-linux-6.10.2-dev/lib/modules/6.10.2/source/Makefile:240: __sub-make] Error 2
       > make[1]: Leaving directory '/nix/store/4pi3spm0m1w4hyr659wkpxmf9cj7b6zr-linux-6.10.2-dev/lib/modules/6.10.2/build'
       > make: *** [Makefile:102: ena.ko] Error 2
       For full logs, run 'nix log /nix/store/j23sjz0wq8gq3y9p8a6082r0nvg9qnyi-ena-2.12.3-6.10.2.drv'.
       ```

akiyano, 2024-07-30:
Hi @arianvp,
Please try this patch 
[0001-workaround-patch-for-kernel-6.10.patch](https://github.com/user-attachments/files/16423057/0001-workaround-patch-for-kernel-6.10.patch)

lano1106, 2024-08-01:
I am just curious. What is the benefit or the reason to use the github version of the driver vs the one that is in intree?

are they different?

the one in drivers/net/ethernet/amazon/ena works out of the box with 6.10.2 for me...

akiyano, 2024-08-02:
Hi @lano1106, 

The driver that is upstream works perfectly well.
But if you wish to have the latest features available in ENA, they are first pushed to the github driver and arrive in the upstream driver later on.

lano1106, 2024-08-02:
oh cool... How can someone know what is in github version not yet in tree?

the commit history?

JonKohler, 2024-09-04:
Heads up @arianvp - more breakage inbound for 6.11 in case you're looking to go up to date: https://github.com/amzn/amzn-drivers/issues/321

davidarinzon, 2024-09-10:
Hi @arianvp 

This issue is expected to be resolved with https://github.com/amzn/amzn-drivers/releases/tag/ena_linux_2.13.0
Please let us know if you experience more issues.

davidarinzon, 2024-09-26:
Resolving the ticket, please let us know if you face any issues.

closed 2024-09-26 by davidarinzon