amzn/amzn-drivers #316: [Support]: need help to remove SOFTIRQ_NET_TX

lano1106, 2024-08-01:
### Preliminary Actions

- [X] I have searched the existing [issues](https://github.com/amzn/amzn-drivers/issues) and didn't find a duplicate.
- [X] I have followed the [AWS official troubleshoot documentation](https://docs.aws.amazon.com/AWSEC2/latest/UserGuide/troubleshooting-ena.html).
- [X] I have followed the driver [readme](https://github.com/amzn/amzn-drivers) and [best practices](https://github.com/amzn/amzn-drivers/blob/master/kernel/linux/ena/ENA_Linux_Best_Practices.rst).

### Driver Type

Linux kernel driver for Elastic Network Adapter (ENA)

### Driver Tag/Commit

kernel 6.9.10-2-ec2 (recompiled by myself from kernel.org git)

### Custom Code

Yes

### OS Platform and Distribution

kernel 6.10.2

$ ethtool -i enp39s0
driver: ena
version: 6.10.2-1-ec2
firmware-version: 
expansion-rom-version: 
bus-info: 0000:27:00.0
supports-statistics: yes
supports-test: no
supports-eeprom-access: no
supports-register-dump: no
supports-priv-flags: no

### Support request

I have been using napi_busy_poll to successfully eliminate most of the ENA driver generated interrupts.

with the appropriate napi_defer_hard_irqs and gro_flush_timeout settings, I have eliminated 99.9% of the SOFTIRQ_NET_RX interrupts.

```
echo 5000 > /sys/class/net/enp39s0/napi_defer_hard_irqs
# gro_flush_timeout unit is nanoseconds
echo 100000 > /sys/class/net/enp39s0/gro_flush_timeout
```
I still have few SOFTIRQ_NET_TX interrupts:

```
     iou-sqp-695-812     [000] ..s1.   988.213106: softirq_entry: vec=2 [action=NET_TX]
     iou-sqp-695-812     [000] ..s..   988.213203: softirq_entry: vec=2 [action=NET_TX]
     iou-sqp-695-812     [000] ..s..   994.213887: softirq_entry: vec=2 [action=NET_TX]
     iou-sqp-695-812     [000] ..s..  1000.214316: softirq_entry: vec=2 [action=NET_TX]
     iou-sqp-695-812     [000] ..s..  1006.214459: softirq_entry: vec=2 [action=NET_TX]
     iou-sqp-695-812     [000] ..s..  1024.214373: softirq_entry: vec=2 [action=NET_TX]
     iou-sqp-695-812     [000] ..s..  1030.214116: softirq_entry: vec=2 [action=NET_TX]
     iou-sqp-695-812     [000] ..s..  1030.214148: softirq_entry: vec=2 [action=NET_TX]
     iou-sqp-695-812     [000] ..s1.  1051.213160: softirq_entry: vec=2 [action=NET_TX]
     iou-sqp-695-812     [000] ..s..  1069.213802: softirq_entry: vec=2 [action=NET_TX]
     iou-sqp-695-812     [000] ..s..  1072.214293: softirq_entry: vec=2 [action=NET_TX]
     iou-sqp-695-812     [000] ..s1.  1075.213749: softirq_entry: vec=2 [action=NET_TX]
     iou-sqp-695-812     [000] ..s1.  1078.214634: softirq_entry: vec=2 [action=NET_TX]
     iou-sqp-695-812     [000] ..s..  1123.214588: softirq_entry: vec=2 [action=NET_TX]
     iou-sqp-695-812     [000] ..s1.  1126.214508: softirq_entry: vec=2 [action=NET_TX]
     iou-sqp-695-812     [000] ..s1.  1129.214314: softirq_entry: vec=2 [action=NET_TX]
     iou-sqp-695-812     [000] ..s1.  1140.127908: softirq_entry: vec=2 [action=NET_TX]
     iou-sqp-695-812     [000] ..s1.  1141.213451: softirq_entry: vec=2 [action=NET_TX]
     iou-sqp-695-812     [000] ..s1.  1141.213481: softirq_entry: vec=2 [action=NET_TX]
     iou-sqp-695-812     [000] ..s..  1153.213411: softirq_entry: vec=2 [action=NET_TX]
     iou-sqp-695-812     [000] ..s1.  1155.127942: softirq_entry: vec=2 [action=NET_TX]
     iou-sqp-695-812     [000] ..s1.  1160.627952: softirq_entry: vec=2 [action=NET_TX]
     iou-sqp-695-812     [000] ..s..  1171.213344: softirq_entry: vec=2 [action=NET_TX]
     iou-sqp-695-812     [000] ..s..  1174.213433: softirq_entry: vec=2 [action=NET_TX]
     iou-sqp-695-812     [000] ..s1.  1177.214668: softirq_entry: vec=2 [action=NET_TX]
     iou-sqp-695-812     [000] ..s1.  1183.214315: softirq_entry: vec=2 [action=NET_TX]
     iou-sqp-695-812     [000] ..s1.  1186.214743: softirq_entry: vec=2 [action=NET_TX]
```

so far the changes that I have made are:

1. recompile kernel with CONFIG_BQL is not set
(this can set __QUEUE_STATE_STACK_XOFF in netdev_tx_sent_queue() which will later schedule a NET_TX call from netdev_tx_completed_queue() that is called by ena_clean_tx_irq())
2. Change value of  ENA_TX_POLL_BUDGET_DIVIDER from 4 to 1

none of these 2 changes succeeded in fully removing NET_TX softirqs... (but I think that are much less numerous).
I am assuming that the only place where the ENA driver can raise NET_TX interrupt is from ena_clean_tx_irq(). Is my assumption correct or I missed and important detail?
is the ethtool -C tx-usecs settings has any incidence on the frequency at which NET_TX is raised?

Also, is there any parameter settings that can manipulate the ena-mgmnt interrupt frequency?

I did quickly read the code in drivers/net/ethernet/amazon/ena/ena_com.c it does some sort of polling about some HW queues but I did find much more info than that on the topic...

lano1106, 2024-08-02:
it for doing NAPI busy polling and make sure that the network i/o latency is at its lowest possible.

depending on the execution time of the interrupt this can add 10-20 uSec latency in whatever is running on that CPU.

that being said, I did backoff a little bit... I did reput CONFIG_BQL back. I did experience a regression in the latency numbers without it...

akiyano, 2024-08-03:
If you wish to work without interrupts we recommend using DPDK.

lano1106, 2024-08-03:
Ok. I'll investigate this option...

Otherwise, I am doing not bad:
```
 65:       8362          0          0          2  PCI-MSIX-0000:27:00.0   0-edge      ena-mgmnt@pci:0000:27:00.0
 66:        261          0          0          0  PCI-MSIX-0000:27:00.0   1-edge      enp39s0-Tx-Rx-0
 67:          0        163          0          1  PCI-MSIX-0000:27:00.0   2-edge    
 68:          1          0        159          0  PCI-MSIX-0000:27:00.0   3-edge    
 69:          0          0          0        134  PCI-MSIX-0000:27:00.0   4-edge    
NMI:          0          0          0          0   Non-maskable interrupts
LOC:    1544473         58         62    1131674   Local timer interrupts
SPU:          0          0          0          0   Spurious interrupts
PMI:          0          0          0          0   Performance monitoring interrupts
IWI:          0         83        110          0   IRQ work interrupts
RTR:          0          0          0          0   APIC ICR read retries
RES:          3          4          3          3   Rescheduling interrupts
CAL:       5408        919       1481   13832208   Function call interrupts
```

and this for doing that:
https://x.com/lano1106/status/1771345949320737183

but before I close this issue, can you give me some details about the admin polling and what it is for?

lano1106, 2024-08-04:
I have found what I wanted to know:
https://www.kernel.org/doc/html/latest/networking/device_drivers/ethernet/amazon/ena.html

the annoying interrupt that I keep seeing is the keep-alive notification. I guess there is not much that can be done to change its period... that being said, it is not happening where I would like it to be.

One of boot command line param that I have is "irqaffinity=3". However, it is useless if the code does not consult irq_default_affinity mask.

but I am not suggesting that ena_setup_mgmnt_intr() be modified to consider irq_default_affinity. It would not change much in my situation because:
```
static int __init irq_affinity_setup(char *str)
{
	alloc_bootmem_cpumask_var(&irq_default_affinity);
	cpulist_parse(str, irq_default_affinity);
	/*
	 * Set at least the boot cpu. We don't want to end up with
	 * bugreports caused by random commandline masks
	 */
	cpumask_set_cpu(smp_processor_id(), irq_default_affinity);
	return 1;
}
__setup("irqaffinity=", irq_affinity_setup);
```

3 is not the boot CPU.

maybe consider adding a module option to let people override the irq selection. For my own needs, I will just hardcode 3 in the driver code...

lano1106, 2024-08-05:
here is the diff of a prototype of mine. It is tested and works as expected. the only thing missing perhaps to integrate it in the official git repository, it might be to replace the define with a module parameter...

One change that should be integrated no matter what is to replace the call to irq_set_affinity_hint() that is now flagged as deprecated. (A comment in include/linux/interrupt.h say so)

```
diff --git a/drivers/net/ethernet/amazon/ena/ena_netdev.c b/drivers/net/ethernet/amazon/ena/ena_netdev.c
index 184b6e6cbed4..56ce81815266 100644
--- a/drivers/net/ethernet/amazon/ena/ena_netdev.c
+++ b/drivers/net/ethernet/amazon/ena/ena_netdev.c
@@ -34,6 +34,8 @@ MODULE_LICENSE("GPL");
 #define DEFAULT_MSG_ENABLE (NETIF_MSG_DRV | NETIF_MSG_PROBE | NETIF_MSG_IFUP | \
 		NETIF_MSG_IFDOWN | NETIF_MSG_TX_ERR | NETIF_MSG_RX_ERR)
 
+#define ENA_ADMIN_CPU 3
+
 static struct ena_aenq_handlers aenq_handlers;
 
 static struct workqueue_struct *ena_wq;
@@ -1618,7 +1620,7 @@ static void ena_setup_mgmnt_intr(struct ena_adapter *adapter)
 	adapter->irq_tbl[ENA_MGMNT_IRQ_IDX].data = adapter;
 	adapter->irq_tbl[ENA_MGMNT_IRQ_IDX].vector =
 		pci_irq_vector(adapter->pdev, ENA_MGMNT_IRQ_IDX);
-	cpu = cpumask_first(cpu_online_mask);
+	cpu = ENA_ADMIN_CPU;
 	adapter->irq_tbl[ENA_MGMNT_IRQ_IDX].cpu = cpu;
 	cpumask_set_cpu(cpu,
 			&adapter->irq_tbl[ENA_MGMNT_IRQ_IDX].affinity_hint_mask);
@@ -1669,7 +1671,7 @@ static int ena_request_mgmnt_irq(struct ena_adapter *adapter)
 		  "Set affinity hint of mgmnt irq.to 0x%lx (irq vector: %d)\n",
 		  irq->affinity_hint_mask.bits[0], irq->vector);
 
-	irq_set_affinity_hint(irq->vector, &irq->affinity_hint_mask);
+	irq_set_affinity_and_hint(irq->vector, &irq->affinity_hint_mask);
 
 	return rc;
 }
@@ -3345,7 +3347,8 @@ static int ena_restore_device(struct ena_adapter *adapter)
 	if (test_bit(ENA_FLAG_LINK_UP, &adapter->flags))
 		netif_carrier_on(adapter->netdev);
 
-	mod_timer(&adapter->timer_service, round_jiffies(jiffies + HZ));
+	adapter->timer_service.expires = round_jiffies(jiffies + HZ);
+	add_timer_on(&adapter->timer_service, ENA_ADMIN_CPU);
 	adapter->last_keep_alive_jiffies = jiffies;
 
 	return rc;
@@ -3708,7 +3711,8 @@ static void ena_timer_service(struct timer_list *t)
 	}
 
 	/* Reset the timer */
-	mod_timer(&adapter->timer_service, round_jiffies(jiffies + HZ));
+	adapter->timer_service.expires = round_jiffies(jiffies + HZ);
+	add_timer_on(&adapter->timer_service, ENA_ADMIN_CPU);
 }
 
 static u32 ena_calc_max_io_queue_num(struct pci_dev *pdev,
@@ -4049,7 +4053,8 @@ static int ena_probe(struct pci_dev *pdev, const struct pci_device_id *ent)
 	ena_update_hints(adapter, &get_feat_ctx.hw_hints);
 
 	timer_setup(&adapter->timer_service, ena_timer_service, 0);
-	mod_timer(&adapter->timer_service, round_jiffies(jiffies + HZ));
+	adapter->timer_service.expires = round_jiffies(jiffies + HZ);
+	add_timer_on(&adapter->timer_service, ENA_ADMIN_CPU);
 
 	dev_info(&pdev->dev,
 		 "%s found at mem %lx, mac addr %pM\n",

```

davidarinzon, 2024-08-05:
Hi @lano1106 

If you're trying to minimize latency, there are recommendations on how to do that in our best practices page (https://github.com/amzn/amzn-drivers/blob/master/kernel/linux/ena/ENA_Linux_Best_Practices.rst).

You may explore further solutions to reduce latency, DPDK is an option, as suggested in one of the replies. Another option would be AF_XDP which utilized zero-copy and by that reduces latency. AF_XDP is expected to be released soon officially (one can experiment with [this patch](https://github.com/amzn/amzn-drivers/issues/221#issuecomment-2025188832) until it's released.

You can change the affinity of the management interrupt manually, the name of the IRQ is https://github.com/torvalds/linux/blob/master/drivers/net/ethernet/amazon/ena/ena_netdev.c#L1614

lano1106, 2024-08-05:
Changing the interrupt and the timer affinity manually is what my patch is doing!

thx for your help!

closed 2024-08-05 by lano1106

davidarinzon, 2024-08-05:
Hey @lano1106 

I'll just note (or rather rephrase) that I meant that you can change the affinity of the interrupt[s] not (only) through code changes but rather through shell commands (for example, https://docs.kernel.org/core-api/irq/irq-affinity.html). This, of course, only handles the admin interrupt (Which also has the keep alive asynchronous events). The timer service, as you already know through your changes, we let the kernel decide.

Thanks!

lano1106, 2024-08-06:
good point. I have tried to change a nvme interrupt affinity through the /proc fs and it did not work.

maybe it did failed because that interrupt is in PCI MSIX irq domain (iow managed)...