amzn/amzn-drivers #317: [Support]: Outdated ENA PMD driver

Binary-Vanguard-12138, 2024-08-05:
### Preliminary Actions

- [X] I have searched the existing [issues](https://github.com/amzn/amzn-drivers/issues) and didn't find a duplicate.
- [X] I have followed the [AWS official troubleshoot documentation](https://docs.aws.amazon.com/AWSEC2/latest/UserGuide/troubleshooting-ena.html).
- [X] I have followed the driver [readme](https://github.com/amzn/amzn-drivers) and [best practices](https://github.com/amzn/amzn-drivers/blob/master/kernel/linux/ena/ENA_Linux_Best_Practices.rst).

### Driver Type

DPDK PMD for Elastic Network Adapter (ENA)

### Driver Tag/Commit

ena_linux_2.12.3

### Custom Code

No

### OS Platform and Distribution

4.14.336-257.568.amzn2.x86_64

### Support request

The ENA PMD driver for DPDK is outdated.
https://github.com/DPDK/dpdk/commit/95eaa71c66eadb6a6924cc5241e02d3dcc58217c
In the commit to the DPDK repository above, the version of the ENA PMD driver has been upgraded to 2.10.0.
But in this repository, it is still 2.8.0.

shaibran, 2024-08-06:
Thank you for contacting us. We will be updating the GitHub repository within the next few days to include the latest driver version and an updated guide detailing all the new features.

Binary-Vanguard-12138, 2024-12-01:
@shaibran Any updates on this issue?
I thought it was easy to resolve this issue.

shaibran, 2025-02-03:
Update: The issue has been resolved. 

The latest driver version 2.11.0, including a follow-up [bug fix](https://patches.dpdk.org/project/dpdk/patch/20241212152455.510-1-shaibran@amazon.com/), has been updated under the [ENA repository](https://github.com/amzn/amzn-drivers/tree/master/userspace/dpdk/ena).