amzn/amzn-drivers #318: [Support]: Performance issue of ENA PMD driver for DPDK

Binary-Vanguard-12138, 2024-08-05:
### Preliminary Actions

- [X] I have searched the existing [issues](https://github.com/amzn/amzn-drivers/issues) and didn't find a duplicate.
- [X] I have followed the [AWS official troubleshoot documentation](https://docs.aws.amazon.com/AWSEC2/latest/UserGuide/troubleshooting-ena.html).
- [X] I have followed the driver [readme](https://github.com/amzn/amzn-drivers) and [best practices](https://github.com/amzn/amzn-drivers/blob/master/kernel/linux/ena/ENA_Linux_Best_Practices.rst).

### Driver Type

DPDK PMD for Elastic Network Adapter (ENA)

### Driver Tag/Commit

ena_linux_2.12.3

### Custom Code

No

### OS Platform and Distribution

4.14.336-257.568.amzn2.x86_64

### Support request

Probably duplicates with #45 , but I could not find answers in the issue.
I did some benchmark tests on AWS `c6in.metal` instances.
I tried to send UDP packets with 32 bytes of payload and used different numbers of TX rings like 8, 16, 32, etc.
Each TX ring had a dedicated CPU logical core.
Firstly, I used DPDK 23.11.1 with ENA PMD driver 2.8.0 and was able to send 8.88 million packets per second with 16 TX rings.
But when I increased the number of TX rings to 32, the total TX packets only increased to 9.17Mpps.

Next, I used DPDK 24.07 with ENA PMD driver 2.10.0, and this time the performance decreased by about half.
I was able to send 4.55 million packets per second with 16 TX rings, and 4.599 million packets per second with 32 TX rings.

When I used kernel driver instead of DPDK PMD driver, I was able to send exactly 20 million packets per second (which seems to be a limitation of AWS instance) using 64 CPU cores.

Not sure why this happens and also not sure if I need to open this issue here or in DPDK repository.

shaibran, 2024-08-06:
Thank you for reaching out to us.
We will conduct our benchmark test on c6in.metal internally and provide an update accordingly.

shaibran, 2024-08-06:
I apologize for overlooking that you are using the c6in.metal instance which is built with [Nitro v4 ](https://docs.aws.amazon.com/ec2/latest/instancetypes/ec2-nitro-instances.html) system which requires to map memory BAR of the ENA as [write combined](https://github.com/amzn/amzn-drivers/tree/master/userspace/dpdk#6-vfio-pci-and-igb_uio), otherwise, performance may be degraded.
Please refer to our [guide](https://github.com/amzn/amzn-drivers/tree/master/userspace/dpdk#62-instructions-for-using-igb_uio-and-vfio-pci) which includes instructions to enable WC. If this does not resolve your issue, please provide your instance ID and the test timeframe (UTC) so we can inspect our logs.

Binary-Vanguard-12138, 2024-08-06:
> I apologize for overlooking that you are using the c6in.metal instance which is built with [Nitro v4 ](https://docs.aws.amazon.com/ec2/latest/instancetypes/ec2-nitro-instances.html) system which [...]

I am pretty sure that I enabled WC.
Can you let me know how I can get my instance ID?
Sorry, I am just a developer who is using the instances that are provided by devops team.

Binary-Vanguard-12138, 2024-08-07:
> > Do you want me to send burst packets with different configurations and write the timeframe?
> 
> Exactly, please provide the test start time in UTC so we can inspect ec2 internal logs during your test. In addition:
> 
> 1. please share which UIO driver you use and the command sequence you used to bring it up and bind to it.
>    [metal ](https://github.com/amzn/amzn-drivers/tree/master/userspace/dpdk#63-note-about-usage-on-metal-instances)instances supports IOMMU, but you may need to disable it in case you are working with `igb_uio ` or `generic_uio_driver` which do not support it.
> 2. please see [guide ](https://github.com/amzn/amzn-drivers/tree/master/userspace/dpdk#623-verification-of-the-write-combined-memory-mapping)for verifying that the memory was mapped as WC.

1. Basically I followed the instructions in this [link](https://doc.dpdk.org/dts/gsg/usr_guide/igb_uio.html).
Here are the commands that I used.
```
cd ~/dpdk_23_11_1/
git clone git://dpdk.org/dpdk-kmods
cd ~/dpdk_23_11_1/dpdk-kmods/linux/igb_uio/
make

sudo modprobe uio
sudo rmmod igb_uio
sudo insmod ./igb_uio.ko wc_enabled=1

cd ~/dpdk_23_11_1/dpdk-stable-23.11.1/usertools/
sudo python3 dpdk-devbind.py --status
sudo python3 dpdk-devbind.py --unbind 0000:00:07.0
sudo python3 dpdk-devbind.py --bind=igb_uio 0000:00:07.0
```
I know when using `vfio_pcio`, we need to enable IOMMU following the instructions in this [link](https://github.com/amzn/amzn-drivers/tree/master/userspace/dpdk#631-x86_64-eg-c5metal-i3metal).
When testing with `igb_uio`, I basically reverted the changes in `/etc/default/grub`, i.e. removed `iommu=1 intel_iommu=on` from the file, and then ran `grub2-mkconfig`, finally rebooted, so it is supposed to disable IOMMU.
2. I also verified that I enabled WC using the guide you provided. I will post the screenshot eventually.

Binary-Vanguard-12138, 2024-08-07:
@shaibran 
Here is the WC check result on `c6in.metal` instance, it looks like WC is not enabled properly.
```
[root@ip-172-31-41-20 ec2-user]# lspci -v -s 0000:09:00.0
09:00.0 Ethernet controller: Amazon.com, Inc. Elastic Network Adapter (ENA)
        Subsystem: Amazon.com, Inc. Elastic Network Adapter (ENA)
        Physical Slot: 2
        Flags: bus master, fast devsel, latency 0, NUMA node 0
        Memory at 9d202000 (32-bit, non-prefetchable) [size=8K]
        Memory at 9d200000 (32-bit, non-prefetchable) [size=8K]
        Memory at 21fffc000000 (64-bit, prefetchable) [size=4M]
        Capabilities: [40] Power Management version 3
        Capabilities: [70] Express Endpoint, MSI 00
        Capabilities: [b0] MSI-X: Enable+ Count=132 Masked-
        Capabilities: [100] #19
        Capabilities: [150] Transaction Processing Hints
        Kernel driver in use: igb_uio
        Kernel modules: ena

[root@ip-172-31-41-20 ec2-user]# cat /sys/kernel/debug/x86/pat_memtype_list | grep 21fffc000000
uncached-minus @ 0x21fffc000000-0x21fffc400000
uncached-minus @ 0x21fffc000000-0x21fffc400000
```

I sent some packets with `igb_uio` enabled 16 TX lcores from about `Wed Aug  7 14:36:33 UTC 2024` to `Wed Aug  7 14:37:33 UTC 2024`, sent 273365929 packets in total.

Binary-Vanguard-12138, 2024-08-07:
@shaibran What could be the reason why WC is not enabled correctly?

Binary-Vanguard-12138, 2024-08-07:
Tried with DPDK 23.11.1, sent 532783050 packets using 16 TX lcores from `Wed Aug  7 14:51:40 UTC 2024` to `Wed Aug  7 14:52:40 UTC 2024`.
WC is same with DPDK 24.07, not enabled properly.
```
[root@ip-172-31-41-20 ec2-user]# lspci -v -s 0000:09:00.0
09:00.0 Ethernet controller: Amazon.com, Inc. Elastic Network Adapter (ENA)
        Subsystem: Amazon.com, Inc. Elastic Network Adapter (ENA)
        Physical Slot: 2
        Flags: bus master, fast devsel, latency 0, NUMA node 0
        Memory at 9d202000 (32-bit, non-prefetchable) [size=8K]
        Memory at 9d200000 (32-bit, non-prefetchable) [size=8K]
        Memory at 21fffc000000 (64-bit, prefetchable) [size=4M]
        Capabilities: [40] Power Management version 3
        Capabilities: [70] Express Endpoint, MSI 00
        Capabilities: [b0] MSI-X: Enable+ Count=132 Masked-
        Capabilities: [100] #19
        Capabilities: [150] Transaction Processing Hints
        Kernel driver in use: igb_uio
        Kernel modules: ena

[root@ip-172-31-41-20 ec2-user]# cat /sys/kernel/debug/x86/pat_memtype_list | grep 21fffc000000
uncached-minus @ 0x21fffc000000-0x21fffc400000
uncached-minus @ 0x21fffc000000-0x21fffc400000
```

Binary-Vanguard-12138, 2024-08-07:
Oops, sorry, but I was using wrong command.
I should use `sudo insmod ./igb_uio.ko wc_activate=1` instead of `wc_enabled=1`.
Will try again with this option.

Binary-Vanguard-12138, 2024-08-07:
I enabled WC properly this time.
```
[root@ip-172-31-41-20 ec2-user]# lspci -v -s 0000:09:00.0
09:00.0 Ethernet controller: Amazon.com, Inc. Elastic Network Adapter (ENA)
        Subsystem: Amazon.com, Inc. Elastic Network Adapter (ENA)
        Physical Slot: 2
        Flags: fast devsel, NUMA node 0
        Memory at 9d202000 (32-bit, non-prefetchable) [size=8K]
        Memory at 9d200000 (32-bit, non-prefetchable) [size=8K]
        Memory at 21fffc000000 (64-bit, prefetchable) [size=4M]
        Capabilities: [40] Power Management version 3
        Capabilities: [70] Express Endpoint, MSI 00
        Capabilities: [b0] MSI-X: Enable- Count=132 Masked-
        Capabilities: [100] #19
        Capabilities: [150] Transaction Processing Hints
        Kernel driver in use: igb_uio
        Kernel modules: ena
[root@ip-172-31-41-20 ec2-user]# cat /sys/kernel/debug/x86/pat_memtype_list | grep 21fffc000000
write-combining @ 0x21fffc000000-0x21fffc400000
```
The TX packets per second improved much, was able to send 17.971 Mpps with 16 TX lcores.
The timestamp is from `Wed Aug  7 15:16:08 UTC 2024` to `Wed Aug  7 15:17:08 UTC 2024`.

However, I could not send 20Mpps (which I believe is a limit of the AWS instance) in any case which made me unhappy.
I was able to send basically same amount with only 8 TX lcores as well, meaning that the increase of TX lcores does not increase the total performance for some reason.

Binary-Vanguard-12138, 2024-08-07:
Also, there is a big difference when I enabled/disabled UDP port cycling.
When I sent from the fixed UDP source port to the fixed UDP destination port, the performance decreased a lot.
When I sent from continuously changing UDP source port to changing UDP destination port, the performance was much better than fixed UDP port.
I was wondering why. @shaibran

Binary-Vanguard-12138, 2024-08-07:
Another timeframe: from `Wed Aug  7 15:30:17 UTC 2024` to `Wed Aug  7 15:31:17 UTC 2024`.
Sent with 10 TX lcores, sent 1095894515 packets in total, which is equivalent with 18.264Mpps.
This result is almost same with when I used vfio-pci module.

Binary-Vanguard-12138, 2024-08-07:
Can I ask what is the optimized configuration to get the maximum pps for TX? @shaibran 
In the `c6in.metal` instance, I have 128 virtual cores on 2 NUMA nodes. Each NUMA node has one physical NIC attached.
I also know there are 32 TX and RX rings for each NIC.
During some experiments, I found that we could achieve the limit i.e. 20Mpps when I was using 32 TX lcores, and each TX lcore was sending packets from a fixed UDP port to a fixed UDP port. (But the destination port for each TX ring was different.)
But with 32 TX cores, when I added UDP port cycling for each packet, i.e. increasing the destination UDP port for each packet in each TX thread, the performance decreased a lot (lower than 15Mpps).

shaibran, 2024-08-08:
As [documented](https://docs.aws.amazon.com/AWSEC2/latest/UserGuide/monitoring-network-performance-ena.html), each EC2 instance has a maximum PPS performance based on its type and size. The metrics that AWS provides can be queried using [`aws ec2 describe-instance-types`](https://docs.aws.amazon.com/AWSEC2/latest/UserGuide/ec2-instance-network-bandwidth.html), which include only bandwidth.

We tested the c6in metal instance and found that it met the Key Performance Indicators (KPI). A review of your instance's internal logs did not show any issues.

Matching applications and algorithms to the underlying architecture is beyond the PMD scope, but we can share some best practices:

1. Identify the NUMA socket to which the adapter is attached and use only the lcores on that NUMA. Note that this varies from platform to platform and cannot be assumed to be sequential. Avoid using lcore 0, as it is the primary core used by Linux.
2. Ensure the application handles pushbacks from the device. If the application floods the device, handling the dropped packets will consume CPU resources and lead to performance degradation.
3. Track the instance allowance via xstats (e.g., [`pps_allowance_exceeded`](https://docs.aws.amazon.com/AWSEC2/latest/UserGuide/monitoring-network-performance-ena.html)).

All the best,
Shai

Binary-Vanguard-12138, 2024-08-08:
> As [documented](https://docs.aws.amazon.com/AWSEC2/latest/UserGuide/monitoring-network-performance-ena.html), each EC2 instance has a maximum PPS performance based on its type and size. The metrics [...]

Thank you.
1. I am sure I used only the lcores on the corresponding NUMA nodes. I admit that I used lcore 0, but not sure it would impact the performance too much. I can try to avoid using lcore 0 though.
2. Can you please explain it in more detail? How can I know if there are pushback from the device and how can I avoid them or handle them properly?
3. Can we use `xstats` when I am using DPDK?
You said `each EC2 instance has a maximum PPS performance based on its type and size`, where can I find this value? If I understand correctly, I can only find bandwidth, not PPS limit.

You also mentioned, `We tested the c6in metal instance and found that it met the Key Performance Indicators (KPI)`.
Can you tell me what KPI you met and what kind of test you conducted?
I am particularly interested about how you assigned lcores when using DPDK, and what values you used for TX descriptor numbers, etc and how many PPS you were able to reach.

shaibran, 2024-08-11:
The metrics that AWS provides externally are those available via the [`aws ec2 describe-instance-types`](https://docs.aws.amazon.com/AWSEC2/latest/UserGuide/ec2-instance-network-bandwidth.html) command. Our KPIs are not available for sharing.

All statistics should be retrieved from the application bound to the network interface; please refer to the provided [instructions](https://github.com/amzn/amzn-drivers/tree/master/userspace/dpdk#8-driver-statistics). Developing a DPDK application falls outside the scope of PMD expertise, so we recommend consulting the open-source DPDK documentation and example applications, such as `testpmd`.

Binary-Vanguard-12138, 2024-08-28:
@shaibran I have checked the performance benchmark test results [here](https://fast.dpdk.org/doc/perf/DPDK_23_03_Intel_NIC_performance_report.pdf).
In test results of `Test#10 Zero packet loss test on Intel server of c6in.8xlarge AWS cloud instance`, when the frame size is 500 bytes, the frame rate is 5.4Mpps and the line rate is 44.93%.
There is a note claiming that `Throughput is limited by the AWS instance configuration on ENA NIC driver.`
I wonder what exactly is this limit.
Is it a limit of PPS or BPS or something else?
Also in Test #10 of other similar [test](https://fast.dpdk.org/doc/perf/DPDK_22_11_Intel_NIC_performance_report.pdf), why can't we achieve full line rate even when we send 1518 bytes of frame?
Additionally, which parameter did you use for the TREX packet generator? Were they TCP or UDP? What were the distributions of IP addresses or ports of sent packets?
Thanks in advance.

shaibran, 2024-08-29:
This report was not crafted by AWS, please reach out to the publisher of that report for the technical details.
I am not aware of any limitation as stated there.

shaibran, 2024-09-26:
I am resolving this issue

closed 2024-09-26 by shaibran