amzn/amzn-drivers #322: [Bug]: ptp_clock_info in ena 2.13 is missing .adjfine

gemorin, 2024-09-25:
### Preliminary Actions

- [X] I have searched the existing [issues](https://github.com/amzn/amzn-drivers/issues) and didn't find a duplicate.
- [X] I have followed the [AWS official troubleshoot documentation](https://docs.aws.amazon.com/AWSEC2/latest/UserGuide/troubleshooting-ena.html).
- [X] I have followed the driver [readme](https://github.com/amzn/amzn-drivers) and [best practices](https://github.com/amzn/amzn-drivers/blob/master/kernel/linux/ena/ENA_Linux_Best_Practices.rst).

### Driver Type

Linux kernel driver for Elastic Network Adapter (ENA)

### Driver Tag/Commit

ena 2.13.0

### Custom Code

No

### OS Platform and Distribution

Linux 6.2+ with ena.phc_enable=1

### Bug description

Starting from Linux 6.2 (commit 75ab70ec5cef), `.adjfine` (introduced in 4.10) is required in `struct ptp_clock_info` and will trigger a NULL pointer dereference in (at least) `ptp_clock_adjtime()`

testing this change:
```
--- a/kernel/linux/ena/ena_phc.c
+++ b/kernel/linux/ena/ena_phc.c
@@ -15,6 +15,15 @@ static int ena_phc_adjfreq(struct ptp_cl
 }
 
 #endif /* ENA_PHC_SUPPORT_ADJFREQ */
+
+#ifdef ENA_PHC_SUPPORT_ADJFINE
+static int ena_phc_adjfine(struct ptp_clock_info *clock_info, long delta)
+{
+       return -EOPNOTSUPP;
+}
+
+#endif /* ENA_PHC_SUPPORT_ADJFINE */
+
 static int ena_phc_adjtime(struct ptp_clock_info *clock_info, s64 delta)
 {
        return -EOPNOTSUPP;
@@ -113,6 +122,9 @@ static struct ptp_clock_info ena_ptp_clo
 #ifdef ENA_PHC_SUPPORT_ADJFREQ
        .adjfreq        = ena_phc_adjfreq,
 #endif /* ENA_PHC_SUPPORT_ADJFREQ */
+#ifdef ENA_PHC_SUPPORT_ADJFINE
+       .adjfine        = ena_phc_adjfine,
+#endif /* ENA_PHC_SUPPORT_ADJFINE */
        .adjtime        = ena_phc_adjtime,
 #ifdef ENA_PHC_SUPPORT_GETTIME64
 #ifdef ENA_PHC_SUPPORT_GETTIME64_EXTENDED
--- a/kernel/linux/ena/kcompat.h
+++ b/kernel/linux/ena/kcompat.h
@@ -995,6 +995,10 @@ static inline bool ktime_after(const kti
 #define ptp_clock_register(info, parent) ptp_clock_register(info)
 #endif
 
+#if LINUX_VERSION_CODE >= KERNEL_VERSION(4, 10, 0)
+#define ENA_PHC_SUPPORT_ADJFINE
+#endif /* ENA_PHC_SUPPORT_ADJFINE */
+
 #endif /* CONFIG_PTP_1588_CLOCK */
 
 #if (LINUX_VERSION_CODE < KERNEL_VERSION(3, 19, 0)) && \
 ```

### Reproduction steps

```bash
1. boot a Linux 6.2+ kernel
```

### Expected Behavior

not crash

### Actual Behavior

crash :-)

### Relevant log output

```shell
abbreviated kernel log:

BUG: kernel NULL pointer dereference, address: 0000000000000000
#PF: supervisor instruction fetch in kernel mode
#PF: error_code(0x0010) - not-present page
PGD 149927067 P4D 149927067 PUD 143ccf067 PMD 0 
Oops: 0010 [#1] SMP NOPTI
CPU: 0 PID: 9217 Comm: t Tainted: G           O        #1
Hardware name: Amazon EC2/Not Specified, BIOS 1.0 10/16/2017
RIP: 0010:0x0
Code: Unable to access opcode bytes at 0xffffffffffffffd6.
Call Trace:
 <TASK>
 ptp_clock_adjtime+0xd1/0x1d0
 pc_clock_adjtime+0x5b/0xa0
 __do_sys_clock_adjtime+0x8a/0x110
 __x64_sys_clock_adjtime+0x16/0x20
 x64_sys_call+0x1321/0x2020
 do_syscall_64+0x33/0x80
 entry_SYSCALL_64_after_hwframe+0x78/0xe2
```

### Contact Details

guillaume@morinfr.org

davidarinzon, 2024-09-26:
Thank you for raising this issue, @gemorin 

We will resolve it in the next driver release.
Do you need a patch for this issue in the meanwhile?

davidarinzon, 2024-11-25:
Hi @gemorin 

The issue has been resolved in https://github.com/amzn/amzn-drivers/releases/tag/ena_linux_2.13.1
Please let us know if it works for you

Ichimikichiki, 2024-11-28:
> Hi @gemorin
> 
> The issue has been resolved in https://github.com/amzn/amzn-drivers/releases/tag/ena_linux_2.13.1 Please let us know if it works for you

Not for me. It just created an issue.
```sh
/usr/src/amzn-drivers-2.13.1/kernel/linux/ena/ena_phc.c:121:10: error: ‘struct ptp_clock_info’ has no member named ‘adjfreq’
  121 |         .adjfreq        = ena_phc_adjfreq,
  ```

Ichimikichiki, 2024-11-28:
> Hi @gemorin
> 
> The issue has been resolved in https://github.com/amzn/amzn-drivers/releases/tag/ena_linux_2.13.1 Please let us know if it works for you

```
/usr/src/amzn-drivers-2.13.1/kernel/linux/ena/ena_phc.c:121:10: error: ‘struct ptp_clock_info’ has no member named ‘adjfreq’
  121 |         .adjfreq        = ena_phc_adjfreq,
      |          ^~~~~~~
compilation terminated due to -Wfatal-errors.
```

davidarinzon, 2024-11-28:
> ```
> /usr/src/amzn-drivers-2.13.1/kernel/linux/ena/ena_phc.c:121:10: error: ‘struct ptp_clock_info’ has no member named ‘adjfreq’
>   121 |         .adjfreq        = ena_phc_adjfreq,
>       | [...]

Hi @Ichimikichiki 

Thanks for raising this, can you please share more information? Which distribution are you using?
I would assume that it didn't compile for you before the patch/new version as well.

Ichimikichiki, 2024-11-28:
> Hi @Ichimikichiki
>
> Thanks for raising this, can you please share more information? Which distribution are you using? I would assume that it didn't compile for you before the patch/new version as well.

It compiled fine on 2.13.0

I'm on Rocky Linux 5.14.0-503.15.1.el9_5.x86_64

davidarinzon, 2024-11-28:
Thank you for raising this @Ichimikichiki 
Indeed, seems that there's a compilation issue with RHEL and RL after the backports of the function removal which is slightly different from the originally reported issue here.

Please accept this temporary patch and let us know if it resolves the issue for you.

[0001-linux-ena-Patch-Resolve-compilation-issue-with-adjfr.patch](https://github.com/user-attachments/files/17943893/0001-linux-ena-Patch-Resolve-compilation-issue-with-adjfr.patch)

davidarinzon, 2025-01-02:
Hi,
The fix has been released as part of https://github.com/amzn/amzn-drivers/releases/tag/ena_linux_2.13.2
You're welcome to check it out and let us know if you face any issues

closed 2025-05-04 by davidarinzon