amzn/amzn-drivers #324: [Documentation]: 404 on https://dl.fedoraproject.org/pub/epel/epel-release-latest-7.noarch.rpm

neilmerchant, 2024-10-09:
### Preliminary Actions

- [X] I have searched the existing [issues](https://github.com/amzn/amzn-drivers/issues) and didn't find a duplicate.

### Driver Type

Linux kernel driver for Elastic Network Adapter (ENA)

### Driver Tag/Commit

Latest

### Documentation issue description

Hi

Trying to update RHEL 7.9 server and got to this step:-

# Replace the 7 with the RHEL version you have (For example - for RHEL 9 use 9)
sudo yum install -y https://dl.fedoraproject.org/pub/epel/epel-release-latest-7.noarch.rpm

https://dl.fedoraproject.org/pub/epel/epel-release-latest-7.noarch.rpm no longer exists. Do you have a link that works for this?

Thanks!

### Contact Details

neil.merchant@sas.com

neilmerchant, 2024-10-09:
@davidarinzon Thanks - I managed to find that and in the next step it now needs the link to the dkms RPM instead of installing it out of EPEL. Ticket was to reflect this in the documentation more than reporting a bug with ENA etc.

davidarinzon, 2024-10-10:
@neilmerchant 

We will improve the documentation for the next driver release.
Thanks for pointing this out.

Let me know if you have further questions.

davidarinzon, 2024-11-25:
Hi @neilmerchant 

The documentation has been updated in https://github.com/amzn/amzn-drivers/releases/tag/ena_linux_2.13.1
Please let us know if you have further questions

closed 2025-01-02 by davidarinzon