amzn/amzn-drivers #325: [Support]: ENA driver in Linux kernel source significantly behind (2.1.x)?

FrothyB, 2024-10-11:
### Preliminary Actions

- [X] I have searched the existing [issues](https://github.com/amzn/amzn-drivers/issues) and didn't find a duplicate.
- [X] I have followed the [AWS official troubleshoot documentation](https://docs.aws.amazon.com/AWSEC2/latest/UserGuide/troubleshooting-ena.html).
- [X] I have followed the driver [readme](https://github.com/amzn/amzn-drivers) and [best practices](https://github.com/amzn/amzn-drivers/blob/master/kernel/linux/ena/ENA_Linux_Best_Practices.rst).

### Driver Type

Linux kernel driver for Elastic Network Adapter (ENA)

### Driver Tag/Commit

NA

### Custom Code

No

### OS Platform and Distribution

Linux 6.11.2-cb1.0.lto.fc40.x86_64 #1 SMP PREEMPT_DYNAMIC

### Support request

I'm running fedora 40 which as far as I can understand is using the kernel provided version of ENA.
$ ethtool -i eth0
driver: ena
version: 6.11.2-cb1.0.lto.fc40.x86_64

However, when I look at the version of the driver in kernel source, it appears to be 2.1? 
https://github.com/torvalds/linux/blob/master/drivers/net/ethernet/amazon/ena/ena_netdev.h

Is there any intention to bring this up to date? Is there any remedy besides installing the driver from source?

davidarinzon, 2024-10-11:
Hi @FrothyB 

At a certain point, the kernel maintainers stopped accepting the method of incrementing network drivers (or any drivers AFAIK) versions, therefore, the driver available in the upstream kernel retained the version 2.1.0k.
It does not reflect the real version of the driver, the only way is to look at the commits history to understand which changes and features are available.
If you wish to use the up-to-date ENA Linux driver, you can install the github driver.

FrothyB, 2024-10-11:
Understood, thank you for clarifying that. Poking around the source, it seems that a lot of the recent work for XDP (for example) is not present in the kernel source, so as you say the most up to date version is available here. Could you comment on roughly how long it takes on average for the kernel driver to catch up?

closed 2024-10-11 by FrothyB

davidarinzon, 2024-10-12:
Hi @FrothyB 

I can't commit on a date, we are working hard on catching up and aligning between the github and the upstream driver where possible.
Please stay tuned.