amzn/amzn-drivers #327: [Support]: Can TX queues have different priorities (arbiter mux)?

likewise, 2024-11-02:
### Preliminary Actions

- [X] I have searched the existing [issues](https://github.com/amzn/amzn-drivers/issues) and didn't find a duplicate.
- [X] I have followed the [AWS official troubleshoot documentation](https://docs.aws.amazon.com/AWSEC2/latest/UserGuide/troubleshooting-ena.html).
- [X] I have followed the driver [readme](https://github.com/amzn/amzn-drivers) and [best practices](https://github.com/amzn/amzn-drivers/blob/master/kernel/linux/ena/ENA_Linux_Best_Practices.rst).

### Driver Type

Linux kernel driver for Elastic Network Adapter (ENA)

### Driver Tag/Commit

HEAD

### Custom Code

No

### OS Platform and Distribution

Linux

### Support request

Does the ENA hardware support hardware prioritized arbitration between the transmit queues, such that packet on a higher prioritized queue get sent earlier than packets on lower prioritized queues? Question is for both Linux and/or DPDK drivers to support this, if available by the ENA.

Or does a single queue support packets bypassing to head-of-line?

### Contact Details

leon@sidebranch.com

davidarinzon, 2024-11-04:
Hi @likewise 

ENA does not support prioritization/arbitration between the queues.

likewise, 2024-11-13:
Thanks for confirming.

closed 2024-11-13 by likewise