amzn/amzn-drivers #329: [Support]: Is LLQ only used for the packet headers, and if so, which headers (up to L4?)

likewise, 2024-11-13:
### Preliminary Actions

- [X] I have searched the existing [issues](https://github.com/amzn/amzn-drivers/issues) and didn't find a duplicate.
- [X] I have followed the [AWS official troubleshoot documentation](https://docs.aws.amazon.com/AWSEC2/latest/UserGuide/troubleshooting-ena.html).
- [X] I have followed the driver [readme](https://github.com/amzn/amzn-drivers) and [best practices](https://github.com/amzn/amzn-drivers/blob/master/kernel/linux/ena/ENA_Linux_Best_Practices.rst).

### Driver Type

DPDK PMD for Elastic Network Adapter (ENA)

### Driver Tag/Commit

HEAD

### Custom Code

No

### OS Platform and Distribution

Linux

### Support request

In deciding to use the large setting for LLQ, I would like to understand if the LLQ is used only for the packet header without payload or for the packet header plus potentially part of the payload?

And if packet header only, will it go up to L4 headers or any other layer?

Question is for both Linux and DPDK drivers.

### Contact Details

leon@sidebranch.com

davidarinzon, 2024-11-25:
Hi @likewise 

Do you have any further questions about this topic?

likewise, 2024-11-25:
Thanks for the answer, that is clear. One further question would be where the LLQ vs non-LLQ code paths are in DPDK and the Linux driver, so that we can verify that the packet was pushed to the LLQ by the CPU.  I know I can dig down the code myself, but maybe there is a very obvious function that you already know about that makes the decision. Or is it just using a special type of descriptor?

closed 2024-11-25 by likewise