amzn/amzn-drivers #330: [Bug]: AF_XDP memory corruption (?) during heavy pps load

derlaft, 2024-11-26:
### Preliminary Actions

- [X] I have searched the existing [issues](https://github.com/amzn/amzn-drivers/issues) and didn't find a duplicate.
- [X] I have followed the [AWS official troubleshoot documentation](https://docs.aws.amazon.com/AWSEC2/latest/UserGuide/troubleshooting-ena.html).
- [X] I have followed the driver [readme](https://github.com/amzn/amzn-drivers) and [best practices](https://github.com/amzn/amzn-drivers/blob/master/kernel/linux/ena/ENA_Linux_Best_Practices.rst).

### Driver Type

Linux kernel driver for Elastic Network Adapter (ENA)

### Driver Tag/Commit

ena_linux_2.13.1

### Custom Code

No

### OS Platform and Distribution

6.8.0-1019-aws

### Bug description

With the following setup:

* 1 instance generating traffic (~2.5M packets per second) with [pktgen_sample02_multiqueue.sh](https://github.com/torvalds/linux/blob/master/samples/pktgen/pktgen_sample02_multiqueue.sh) + [patch to set specific PPS rate
](https://gist.github.com/derlaft/d4b65e90d9be8773bee33df41bc3ba0f)
* 1 instance processing this traffic using AF_XDP [using a slightly modified example from xdp-tutorial](https://gist.github.com/derlaft/9c4a7246f534ea7d055996c8c7d83fec). The idea is to just discard each read packet after doing `usleep(100);`.

When zero copy mode is enabled, memory corruption (?) is observed: packets read from RX ring change in realtime. The corruption is observed by checking if ethernet protocol field is somehow set to 0.

### Reproduction steps

```bash
Generator: run packet load on the generating instance:

RATEP=625000 pktgen/pktgen_sample02_multiqueue.sh -d ${TARGET_IP} -m ${TARGET_MAC} -p 30769-33998 -i ens5 -t 4 -n 0
```

Processor: run the modified example in zero-copy mode:
```
sudo ethtool -L ens6 combined 1
sudo ip link set dev ens6 mtu 3200
sudo ip route del default dev ens6
make && sudo ./af_xdp_user -N -z -d ens6 -Q 0 -q 
```
```

### Expected Behavior

Works just like in copy mode, without corrupted data. To run in copy mode: `./af_xdp_user -N -c -d ens6 -Q 0 -q `

### Actual Behavior

```
$ make && sudo ./af_xdp_user -N -z -d ens6 -Q 0 -q 
Memory corruption?
IP packet: 
0200EFFB4FC3024A6D10D20B08004500002E441E0000201121190A5910780A59104300277BF9001A0000BE9BE9553355444F00000000000000000000
The same IP packet after 1s sleep: 
0252EFFB4F0000000000000000004500002ECE1300002011973F0A5910780A591043003979D8001A0000BE9BE955335ECE2E00000000000000000000
```

### Additional Data

```bash
region = eu-west-1
instance_type = c7gn.xlarge
instance_id = i-094f1c95c0171b69e
ethtool -S ens6 = https://gist.github.com/derlaft/83a5fb38b691279aa8da2f7014f5ed05
```

### Contact Details

aleksei.kharlamov@emnify.com

davidarinzon, 2024-11-28:
Thank you for reporting it and sharing the information and the tools for re-creation, @derlaft 
We are looking into it.

derlaft, 2024-12-03:
Huge thanks for the quick fix @ShayAgros, I've tested the patch and it indeed fixes the reported problem. If we encounter any more issues in AF_XDP support, I'll open another issue :)

closed 2024-12-03 by derlaft