amzn/amzn-drivers #333: [Bug]: Incompatibility of enav2-vfio-patch on Ubuntu 24.04

Binary-Vanguard-12138, 2024-12-22:
### Preliminary Actions

- [X] I have searched the existing [issues](https://github.com/amzn/amzn-drivers/issues) and didn't find a duplicate.
- [X] I have followed the [AWS official troubleshoot documentation](https://docs.aws.amazon.com/AWSEC2/latest/UserGuide/troubleshooting-ena.html).
- [X] I have followed the driver [readme](https://github.com/amzn/amzn-drivers) and [best practices](https://github.com/amzn/amzn-drivers/blob/master/kernel/linux/ena/ENA_Linux_Best_Practices.rst).

### Driver Type

DPDK PMD for Elastic Network Adapter (ENA)

### Driver Tag/Commit

ena_linux_2.13.1

### Custom Code

No

### OS Platform and Distribution

Ubuntu 24.04.1 LTS
20-Ubuntu SMP Thu Oct 10 18:14:42 UTC 2024 x86_64 x86_64 x86_64 GNU/Linux

### Bug description

When I tried to run amzn-drivers/userspace/dpdk/enav2-vfio-patch/get-vfio-with-wc.sh on Ubuntu24.04, I got the following error.
```
 linux-aws-6.8.0/virt/kvm/kvm_main.c
W: Download is performed unsandboxed as root as file 'linux-aws_6.8.0-1021.23.dsc' couldn't be accessed by user '_apt'. - pkgAcquire::Run (13: Permission denied)
Done

[2] Patch and build the vfio driver
Apply patch for the write combining to the vfio-pci
Using patch for kernel version 5.15
patching file drivers/vfio/pci/vfio_pci_core.c
Hunk #1 succeeded at 1745 (offset 252 lines).
Hunk #2 FAILED at 1506.
1 out of 2 hunks FAILED -- saving rejects to file drivers/vfio/pci/vfio_pci_core.c.rej
ERROR: Cannot apply patch: /home/ubuntu/dpdk/amzn-drivers/userspace/dpdk/enav2-vfio-patch/patches/linux-5.15-vfio-wc.patch!
```

### Reproduction steps

```bash
1. Install a fresh Ubuntu 24.04.1 OS on AWS `c6in.16xlarge` instance.
2. Download and run the script.

  git clone https://github.com/amzn/amzn-drivers
  cd amzn-drivers/userspace/dpdk/enav2-vfio-patch
  sudo ./get-vfio-with-wc.sh
```
3. Got the error
```
 linux-aws-6.8.0/virt/kvm/guest_memfd.c
 linux-aws-6.8.0/virt/kvm/kvm_main.c
W: Download is performed unsandboxed as root as file 'linux-aws_6.8.0-1021.23.dsc' couldn't be accessed by user '_apt'. - pkgAcquire::Run (13: Permission denied)
Done

[2] Patch and build the vfio driver
Apply patch for the write combining to the vfio-pci
Using patch for kernel version 5.15
patching file drivers/vfio/pci/vfio_pci_core.c
Hunk #1 succeeded at 1745 (offset 252 lines).
Hunk #2 FAILED at 1506.
1 out of 2 hunks FAILED -- saving rejects to file drivers/vfio/pci/vfio_pci_core.c.rej
ERROR: Cannot apply patch: /home/ubuntu/dpdk/amzn-drivers/userspace/dpdk/enav2-vfio-patch/patches/linux-5.15-vfio-wc.patch!
```
```

### Expected Behavior

The patch works without error.

### Actual Behavior

Patch fails with an error.

shaibran, 2024-12-23:
Thanks for informing us, we will look into this

shaibran, 2025-02-03:
Update: The issue has been resolved. 

The [patches section](https://github.com/amzn/amzn-drivers/tree/master/userspace/dpdk/enav2-vfio-patch/patches) has been updated to include a patch for Linux kernel 6.8, which is used in Ubuntu 24.04 LTS. 

Additionally, the [get-vfio-with-wc.sh](https://github.com/amzn/amzn-drivers/blob/master/userspace/dpdk/enav2-vfio-patch/get-vfio-with-wc.sh) script, which automatically patches the kernel, has been modified accordingly.

andyz422, 2025-04-12:
Hi. I'm running m6in.4xlarge on Ubuntu 24.04 and I'm running into the same issue with the updated patch for Linux kernel 6.8:

`[2] Patch and build the vfio driver
Apply patch for the write combining to the vfio-pci
Using patch for kernel version 6.8
(Stripping trailing CRs from patch; use --binary to disable.)
patching file drivers/vfio/pci/vfio_pci_core.c
Hunk #1 succeeded at 1868 (offset 123 lines).
patch unexpectedly ends in middle of line
Hunk #2 FAILED at 1758.
1 out of 2 hunks FAILED -- saving rejects to file drivers/vfio/pci/vfio_pci_core.c.rej
ERROR: Cannot apply patch: /home/ubuntu/amzn-drivers/userspace/dpdk/enav2-vfio-patch/patches/linux-6.8-vfio-wc.patch!`

andyz422, 2025-04-12:
Hi. I'm running m6in.4xlarge on Ubuntu 24.04 and I'm running into the same issue with the updated patch for Linux kernel 6.8:
```shell
[2] Patch and build the vfio driver 
Apply patch for the write combining to the vfio-pci 
Using patch or kernel version 6.8 (Stripping trailing CRs from patch; use --binary to disable.) 
patching file drivers/vfio/pci/vfio_pci_core.c 
Hunk #1 succeeded at 1868 (offset 123 lines). 
patch unexpectedly ends in middle of line 
Hunk #2 FAILED at 1758. 1 out of 2 hunks 
FAILED -- saving rejects to file drivers/vfio/pci/vfio_pci_core.c.rej 
ERROR: Cannot apply patch: /home/ubuntu/amzn-drivers/userspace/dpdk/enav2-vfio-patch/patches/linux-6.8-vfio-wc.patch!

```

shaibran, 2025-04-17:
Thank you [andyz422](https://github.com/andyz422) for bringing the issue to our attention.
We recreated the issue in house and will update soon

shaibran, 2025-04-17:
Ubuntu 24.04 LTS keeps getting kernel updates which caused problem when applying the patch (line mismatch).
Script is now fixed.

Binary-Vanguard-12138, 2025-05-04:
Hi, I am running Ubuntu 24.04.2 LTS on AWS `c6in.metal` instance, and this script seemed to run without any errors.
But when I ran a DPDK application, I was able to see that write verification is not enabled according to this [link](https://github.com/amzn/amzn-drivers/tree/master/userspace/dpdk#623-verification-of-the-write-combined-memory-mapping).

Here are some command outputs from my terminal.
```
ubuntu@ip-10-195-0-16:~/ittradetech-dpdk/C++/DPDK-UDP-Test/lib$ lspci -v -s  0a:00.0
0a:00.0 Ethernet controller: Amazon.com, Inc. Elastic Network Adapter (ENA)
        Subsystem: Amazon.com, Inc. Elastic Network Adapter (ENA)
        Physical Slot: 3
        Flags: fast devsel, NUMA node 0, IOMMU group 20
        Memory at 9d002000 (32-bit, non-prefetchable) [size=8K]
        Memory at 9d000000 (32-bit, non-prefetchable) [size=8K]
        Memory at 21fffb000000 (64-bit, prefetchable) [size=4M]
        Capabilities: <access denied>
        Kernel driver in use: vfio-pci
        Kernel modules: ena

ubuntu@ip-10-195-0-16:~/ittradetech-dpdk/C++/DPDK-UDP-Test/lib$ sudo su
root@ip-10-195-0-16:/home/ubuntu/ittradetech-dpdk/C++/DPDK-UDP-Test/lib# cat /sys/kernel/debug/x86/pat_memtype_list | grep 21fffb000000
root@ip-10-195-0-16:/home/ubuntu/ittradetech-dpdk/C++/DPDK-UDP-Test/lib# cat /sys/kernel/debug/x86/pat_memtype_list | grep 21fffb000000
PAT: [mem 0x000021fffb000000-0x000021fffb400000] uncached-minus
root@ip-10-195-0-16:/home/ubuntu/ittradetech-dpdk/C++/DPDK-UDP-Test/lib# cat /sys/kernel/debug/x86/pat_memtype_list | grep 21fffb000000
PAT: [mem 0x000021fffb000000-0x000021fffb400000] uncached-minus
root@ip-10-195-0-16:/home/ubuntu/ittradetech-dpdk/C++/DPDK-UDP-Test/lib# cat /sys/kernel/debug/x86/pat_memtype_list | grep 21fffb000000
```
As you can see, I saw `uncached-minus` instead of `write-combining` string.
Can you please check what was wrong? @shaibran

shaibran, 2025-05-21:
Thank you for bringing this to our attention, I have reopened the issue.
We were able to reproduce this in-house and issue is under investigation.

shaibran, 2025-05-21:
the patch does not apply correctly, but it does not show an error, so WC is not being enabled as expected.
we will prepare a fix for the short term, and we are considering a different approach for applying the required change going forward.

shaibran, 2025-05-29:
Issue was root caused to a patch format bug. The [patch](https://github.com/amzn/amzn-drivers/blob/master/userspace/dpdk/enav2-vfio-patch/patches/linux-6.8-vfio-wc.patch) and the [script](https://github.com/amzn/amzn-drivers/blob/master/userspace/dpdk/enav2-vfio-patch/get-vfio-with-wc.sh) that triggers it were reworked and published

llqqp, 2025-05-29:
Hi @shaibran , I'm really sorry to bring this issue back. 
Unfortunately, I'm still having issue with the VFIO patch on Ubuntu 24.04 ami-084568db4383264d4, which is documented as test passed in the readme of the patch.
I'm using a brand-new launched EC2 instance and applied the latest patch immediately without any apt update or apt upgrade. The kernel version is still 6.8.0-1024-aws.
However I'm getting the error message "ERROR: Failure during vfio-pci module installation. Prehaps it's not provided as a kernel module!", then the patching job was quit.
Would you mind verifying if you can reproduce this issue on that AMI again? Is there any more information I can provide to help you for the troubleshooting?
Appreciate again for your help!

shaibran, 2025-06-01:
ami-084568db4383264d4 is the Ubuntu 24.04 amd64 image in IAD.
We tested the fix on intel, we will attempt to reproduce with this configuration and update.

shaibran, 2025-06-04:
Hi, support for zst compression was unintentionally removed, which led to the issue.  
A [fix](https://github.com/amzn/amzn-drivers/commit/e3092d5acc66e4d44307434551ecca6ec952412c) has been merged. Apologies for the inconvenience

closed 2025-06-04 by shaibran