amzn/amzn-drivers #334: [Support]: Interrupts seem only delivering to AF_XDP core rather than the core specified by IRQ affinity

YangZhou1997, 2025-01-05:
### Preliminary Actions

- [X] I have searched the existing [issues](https://github.com/amzn/amzn-drivers/issues) and didn't find a duplicate.
- [X] I have followed the [AWS official troubleshoot documentation](https://docs.aws.amazon.com/AWSEC2/latest/UserGuide/troubleshooting-ena.html).
- [X] I have followed the driver [readme](https://github.com/amzn/amzn-drivers) and [best practices](https://github.com/amzn/amzn-drivers/blob/master/kernel/linux/ena/ENA_Linux_Best_Practices.rst).

### Driver Type

Linux kernel driver for Elastic Network Adapter (ENA)

### Driver Tag/Commit

ena_linux_2.13.0

### Custom Code

No

### OS Platform and Distribution

Ubuntu 22.04.5 LTS (GNU/Linux 6.8.0-1015-aws x86_64)

### Support request

Hi AWS driver maintainer, 

I am using AFXDP zero-copy support from AWS ENA driver to send and receive 100Gbps traffic. I configure multiple NICs queues to receive interrupts, and bind each queue's irq affinity to different cores (through `/proc/irq/$IRQ/smp_affinity_list`). I also use another set of cores to run user-space applications (which receive and send packets using AFXDP APIs). 

However, from `htop`, I find that the IRQ cores (ie, cores specified by irq affinity) do not nearly consume any CPU time, while the application cores consume around half of CPU time in kernel (ie, red bars in htop). From `perf`, the application cores are spending significant time on syscalls like `__lib_sendto` and also `net_rx_action`. So it seems that the NIC interrupts are handled by the application cores. 

I am worried that this frequent context switching between user and kernel space causes poor networking performance. For example, when using Mellanox ConnectX-5 100G NIC with mlx5 driver, my AFXDP applications only requires 4 application cores and 4 IRQ cores to saturate dual-directional 200G traffic; on these 4 application cores, nearly no CPU time spent on the kernel, while the 4 IRQ cores are spent significant time in the kernel handling interrupts. When using c5.18xlarge with AWS 100G NIC and ENA driver, I need to use 12 application cores and specify 12 IRQ cores (ie, 12 NIC queues) to saturate only 150G traffic. 

Another short question is: how to determine the NUMA affinity of the ENA NIC? c5.18xlarge has two NUMA nodes, but I got -1 from `/sys/bus/pci/devices/<PCI_device_ID>/numa_node` and `/sys/class/net/<nic_dev>/device/numa_node`. 

Best,
Yang

davidarinzon, 2025-01-06:
Hi @YangZhou1997 

Thank you for raising this issue, we will look into it and provide feedback soon.

YangZhou1997, 2025-01-07:
Thank you, David! 

Another note is: on the c5n.18xlarge instance, using iperf with 32 connection (ie, -P 32 --dualtest) is able to achieve around 190 Gbps bandwidth under 9k MTU; with 3.5k MTU (ie, the maximum MTU of AFXDP), iperf is able to achieve around 183 Gbps. In comparison, AFXDP with 3.5k MTU can only achieve 150 Gps. I suspect there might be some driver issues for AFXPD support. 

Btw, I disable interrupt collapsing by `sudo ethtool -C ${NIC} adaptive-rx off rx-usecs 0 tx-usecs 0`, as I found it typically does not help much. 

Best,
Yang

YangZhou1997, 2025-01-13:
@ShayAgros I clean up the afxdp repo I used to test network bw on c5n.18xlarge: https://github.com/uccl-project/uccl. 

The readme also has a thorough guide on how to run my code. To reproduce the 150Gbps bw, one needs to: 
* spawn two c5n.18xlarge (instead of g4dn.8xlarge in the readme) 
* run up to step 3 in [Getting Started](https://github.com/uccl-project/uccl?tab=readme-ov-file#getting-started)
* using `python setup_all.py --target aws_c5_afxdp` to build

I would appreciate any help that helps debug the performance issue.

ShayAgros, 2025-01-20:
Just an update, I followed the README and currently stuck at the phase of initializing the tests:
> Run UCCL transport tests on VM1:
> cd /opt/uccl && git pull
Edit nodes.txt to only include the two public IPs of the VMs
Build UCCL:
>
 >   python setup_all.py --target aws_g4_afxdp
 >   Keep setup_all.py running

The python setup_all.py keeps throwing exception (more specifically paramiko is complaining that it's not able to connect to the other host) which is also reproducible by running paramiko manually:
```
>>> client.connect("172.31.105.125")
Unknown exception: public_blob
Traceback (most recent call last):
  File "/opt/conda/lib/python3.12/site-packages/paramiko/transport.py", line 2262, in run
    handler(m)
  File "/opt/conda/lib/python3.12/site-packages/paramiko/auth_handler.py", line 394, in _parse_service_accept
    key_type, bits = self._get_key_type_and_bits(self.private_key)
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/conda/lib/python3.12/site-packages/paramiko/auth_handler.py", line 218, in _get_key_type_and_bits
    if key.public_blob:
       ^^^^^^^^^^^^^^^
  File "/opt/conda/lib/python3.12/site-packages/paramiko/agent.py", line 476, in __getattr__
    raise AttributeError(name)
AttributeError: public_blob

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/opt/conda/lib/python3.12/site-packages/paramiko/client.py", line 485, in connect
    self._auth(
  File "/opt/conda/lib/python3.12/site-packages/paramiko/client.py", line 754, in _auth
    self._transport.auth_publickey(username, key)
  File "/opt/conda/lib/python3.12/site-packages/paramiko/transport.py", line 1709, in auth_publickey
    return self.auth_handler.wait_for_response(my_event)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/conda/lib/python3.12/site-packages/paramiko/auth_handler.py", line 248, in wait_for_response
    raise e
  File "/opt/conda/lib/python3.12/site-packages/paramiko/transport.py", line 2262, in run
    handler(m)
  File "/opt/conda/lib/python3.12/site-packages/paramiko/auth_handler.py", line 394, in _parse_service_accept
    key_type, bits = self._get_key_type_and_bits(self.private_key)
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/conda/lib/python3.12/site-packages/paramiko/auth_handler.py", line 218, in _get_key_type_and_bits
    if key.public_blob:
       ^^^^^^^^^^^^^^^
  File "/opt/conda/lib/python3.12/site-packages/paramiko/agent.py", line 476, in __getattr__
    raise AttributeError(name)
AttributeError: public_blob
```

I'm not familiar enough with Paramiko so it'd take me some time to understand what it wants (invoking ssh from the terminal just works w/o any additional parameters).

I'll probably switch to create a similar setup of several queues receiving traffic and redirecting it to a third server to find whether I can reproduce the bandwidth bottleneck mentioned in the ticket

YangZhou1997, 2025-01-21:
Thank you @ShayAgros! I am trying to remove the dependency of paramiko to make it easier to run. 

The setup you mention would be very helpful for the debugging, as it largely simplifies the settings to the most basic part (my code uccl has implemented a bunch of reliable transport components such as loss recovery). Appreciate your efforts! 

Just one note: AWS has per-flow rate limit of 5Gbps, and you may need to use dozens of flows (eg, with different udp ports) to bypass this limit.

ShayAgros, 2025-02-11:
Hi, sorry for the time it took me to build up the setup and run the checks.

Running a simple AF XDP program which creates 16 zero-copy queues that pass every buffer from the receive queue to the send queue of the same channel and running the iperf2 command from the client (udp transmission of 3400 byte packets with 50 streams):
```
iperf -c 172.31.53.16 -l 3400 -b 200G -i 1 -t 5000 -P 50 -u
```

I'm able to reach the following BW on the server side (where the AF XDP program is running) per second:
```
total_tx_xsk_bits: 90.69G
total_rx_bits: 90.68G
```

which is around 181.37Gbps total

Running the same iperf command from the client and the server (as to simulate the bidirectional traffic) I get:
```
total_tx_bits: 96.91G
total_rx_bits: 85.31G
```

which is around 182.12Gbps

Both use cases are not optimized (though I'd argue that the network stack with `iperf` provides better batching support than AF XDP) so it should be possible to reach better results in my opinion. However, I don't see performance degradation in AF XDP.

Let me know if you want to dive deeper into the issue here.

A few words about my setup:
- I used c5n.18xlarge instances, both running Ubuntu 24.04
- I use a separate thread (and core) for every queue processing in the AF XDP program
- I have a separate thread for *each* of the threads above to trigger a wake up of the driver, i.e. call napi (the thread's affinity is the same as the IRQ corresponding to this queue, as recommended in this ticket)
- The MTU on the server is 3498 and I send 3400 byte packets between the instances
- The client side (w/o the AF XDP program) uses all 32 queues, and the AF XDP instance uses 16
- The stats are taken from the driver on the server using the command `ethtool -S [interface]` (I have a simple script which diffs the stats between two consecutive seconds)

ShayAgros, 2025-02-12:
I think I can share the code I used. Can you please contact me at shayagr@amazon.com ?

closed 2025-03-09 by YangZhou1997