amzn/amzn-drivers #335: [Support]: ENA DPDK driver header splitting

YangZhou1997, 2025-01-30:
### Preliminary Actions

- [x] I have searched the existing [issues](https://github.com/amzn/amzn-drivers/issues) and didn't find a duplicate.
- [x] I have followed the [AWS official troubleshoot documentation](https://docs.aws.amazon.com/AWSEC2/latest/UserGuide/troubleshooting-ena.html).
- [x] I have followed the driver [readme](https://github.com/amzn/amzn-drivers) and [best practices](https://github.com/amzn/amzn-drivers/blob/master/kernel/linux/ena/ENA_Linux_Best_Practices.rst).

### Driver Type

DPDK PMD for Elastic Network Adapter (ENA)

### Driver Tag/Commit

ena_linux_2.13.2

### Custom Code

No

### OS Platform and Distribution

None

### Support request

Hi amazon drivers maintainers! 

I wonder if the ENA DPDK driver supports header splitting, say packet header goes to one buffer while packet payload goes to another. Mellanox NIC DPDK drivers supports since CX5, so I wonder if ENA DPDK driver happens to support it. 

Best,
Yang

shaibran, 2025-02-03:
Hello [YangZhou1997](https://github.com/YangZhou1997),

If you're referring to `Buffer split on Tx` this functionality is supported by the ENA DPDK driver. The application can send a packet using multiple mbuf descriptors, with the header in the first mbuf and the payload in the subsequent mbufs. However, if the number of mbuf descriptors exceeds the maximum supported by the device, the driver will perform linearization, which means all data will be copied into a single mbuf.

If you're asking about the `Buffer split on Rx` feature as mentioned in [this DPDK guide](https://doc.dpdk.org/guides/nics/overview.html), the ENA driver does not support this feature.

Let me know if you need further clarification,
Shai

closed 2025-02-04 by YangZhou1997

shaibran, 2025-02-23:
Dear [YangZhou1997](https://github.com/YangZhou1997),

I hope this message finds you well. I was recently informed about the potential consideration of the header split capability. Although the issue has been closed, I would like to ensure that I fully understand your use case to assess its relevance.

Please feel free to reach out to me directly via email at shaibran@amazon.com if you would like to discuss this further.