amzn/amzn-drivers #336: [Bug]: Frequent ENA driver resets due to "napi handler hasn't been called for a long time"

PaulFurtado, 2025-02-25:
### Preliminary Actions

- [x] I have searched the existing [issues](https://github.com/amzn/amzn-drivers/issues) and didn't find a duplicate.
- [x] I have followed the [AWS official troubleshoot documentation](https://docs.aws.amazon.com/AWSEC2/latest/UserGuide/troubleshooting-ena.html).
- [x] I have followed the driver [readme](https://github.com/amzn/amzn-drivers) and [best practices](https://github.com/amzn/amzn-drivers/blob/master/kernel/linux/ena/ENA_Linux_Best_Practices.rst).

### Driver Type

Linux kernel driver for Elastic Network Adapter (ENA)

### Driver Tag/Commit

v2.13.2g

### Custom Code

No

### OS Platform and Distribution

We run AlmaLinux 9, but we compile our kernel from the AL2023 kernel RPM spec. So `uname -a` is:
```
Linux ip-172-18-78-242 6.1.127-hs95.el9.x86_64 #1 SMP PREEMPT_DYNAMIC Wed Feb 19 00:18:36 UTC 2025 x86_64 x86_64 x86_64 GNU/Linux
```
but the kernel is effectively the same as the current latest kernel available on AL2023 with no further patches.

### Bug description

On our kubernetes nodes, for probably 12+ months, we have seen very frequent ENA driver resets with logs like:
```
ena 0000:90:00.0 eth0: NETDEV WATCHDOG: CPU: 56: transmit queue 11 timed out 5340 ms
ena 0000:90:00.0 eth0: TX q 11 is paused for too long (threshold 5000000). Time since last napi 8270000 usec. napi scheduled: 1. msecs since last interrupt: 8260
ena 0000:90:00.0 eth0: napi handler hasn't been called for a long time but is scheduled
```
another example:
```
ena 0000:8b:00.0 eth0: NETDEV WATCHDOG: CPU: 25: transmit queue 6 timed out 5270 ms
ena 0000:8b:00.0 eth0: TX q 6 is paused for too long (threshold 5000000). Time since last napi 8190000 usec. napi scheduled: 1. msecs since last interrupt: 8190
ena 0000:8b:00.0 eth0: napi handler hasn't been called for a long time but is scheduled
```
Full snippets for those two examples here: https://gist.github.com/PaulFurtado/4397e35a2ce92d93923a0fd2a2548e46

Both of those instances were `m7i.48xlarge`, however, we run m7i/c7i/m8g in this kubernetes cluster and it happens on all of them, we haven't found any correlation to instance type, and last year we probably still had 5th generation types too that were also seeing this.

These errors are particularly problematic for us because apps that have long-running connections don't recover quickly when the driver resets (mysql replication is the biggest pain point), so every time this happens, we hard-kill the instance which is quite disruptive (but the lesser evil vs leaving connections in a stuck state for many minutes).

If helpful, all of these nodes have 4 ENIs attached to them. eth0 is used by the node itself, and then eth1-3 are given 50 IPs each and mapped into pods using ipvlan.

### Reproduction steps

```bash
We have not been able to come up with a good reproducer for this. We believe it to be something about the workloads on the node that is triggering it, but we haven't been able to isolate which it is.

The kubernetes cluster we see this the most in is our QA environment in our biggest region. It has over 500 nodes in it, and all of them are pretty densely packed with 150 pods each, constituting a wide variety of workloads (MySQL, Zookeeper, Java backend services, hadoop jobs, spark jobs, go applications, python applications, nginx load balancers, etc).

Our prod environments see this far less frequently. This is interesting because those nodes are just as big, are running far fewer workloads, but are driving far more CPU and network usage.

Additionally, this happens the most frequently when we do a rolling replacement of all nodes in the cluster. We typically see 1-5 events per day every day, but on a day where we do a rolling replacement of all nodes, such as today, we saw 10 cases.
```

### Expected Behavior

The ENA driver should not be seeing frequent resets on healthy instances

### Actual Behavior

The ENA driver sees frequent resets on healthy instances

### Additional Data

```bash

```

### Relevant log output

```shell

```

### Contact Details

pfurtado@hubspot.com

ShayAgros, 2025-02-26:
Hi Paul

Thank you for opening the ticket and providing the information.

On first glance it seems like the network processing function (napi) isn't being invoked which might lead to both errors  posted by you.

Can you  open this case through AWS support system please ? This would allow the relevant teams (Including ENA) to investigate the issue, since it might be also related to the infrastructure kubernetes uses

PaulFurtado, 2025-02-26:
Hi @ShayAgros, I'm working with our Account SA (Ramesh Jatiya) to open an internal ticket.

> since it might be also related to the infrastructure kubernetes uses

Regarding this, I just want to clarify that we operate our own kubernetes clusters and do not use EKS or any nonstandard EC2 features. So I think the only thing here would be if it is somehow related to using multiple ENIs, or the interplay between the ipvlan driver and the ena driver. But yes, we can follow up via the ticket.

ShayAgros, 2025-03-05:
Hi, sorry for not replying earlier.

Just throwing a guess here. We often see such issues when there are excessive writes to the serial console of the instance (what is later passed to CloudWatch). Such writes are often done by a cloud-init script configured by customers which by default directs its output to both
- `/var/log/cloud-init-output.log`
- The kernel ring (e.g. by writing directly to `/dev/kmsg` or by using a helper logging command)

In cases cloud-init runs an output-intensive command, the serial console driver ends up entering a kind of a busy-poll passing these messages to the serial connection. Unfortunately it runs with a priority high enough to starve the network driver's polling function. We're currently working on elevating this issue, at least partially.

Just in case this might relate to your current situation, are the resets often accompanied with many prints by cloud-init ? If so, would it be possible to direct cloud-init's commands output to a file ? (e.g. if it runs `yum install clang`, then maybe change it to `yum install clang > /var/log/deps_install`)

PaulFurtado, 2025-03-06:
Hi @ShayAgros 

We don't use cloud-init, we have our own tools for configuring the instance on first boot and it doesn't log to the serial console. However, even if it did log there, this is happening way after boot (like days/weeks/months later). At that point, the only thing logging via kmsg is the kernel itself. Unfortunately, there is a ton of stuff that can trigger lots of logs, such as OOM-kill inside containers, or network interfaces being renamed, etc.

When we were investigating this, it didn't seem like there were many kmsg logs around the time of the incidents, but we can look again. We could also attempt to disable the serial console and see if the problem goes away, that is just unfortunate for debugging. We could use netconsole instead though if it doesn't have the same problem.

What resource is the ENA driver competing with the serial console for? Is it purely CPU on a thread they share? Interrupts? A lock? Something else? I'm wondering if there's any way we could observe that resource so we can track down what is consuming it.

Thanks,
Paul

ShayAgros, 2025-03-07:
Hi,
The contention with the serial driver is on the CPU in case the napi handler and the serial console driver happen to run on the same one. If you don't think it's the case let me look for other reasons that might trigger this driver reset

ShayAgros, 2025-03-10:
Still cannot say that I have a solid theory for why this happens. Do you know perhaps whether your instances have a proactive irq balancer ? Is your workflows includes modifying the IRQ affinity (other than what IRQ balancer does automatically)?

PaulFurtado, 2025-03-10:
Hi @ShayAgros,

We did disable the serial console on one cluster on Friday, and we still had occurrences of this over the weekend, so I think we can rule out serial console.

We do run the standard irqbalance service on our instances, and don't do anything else to manager IRQs. Do you think we should stop using irqbalance? Is there a best practice guide for managing IRQs on modern EC2 instances?

Thanks,
Paul

ShayAgros, 2025-03-17:
The interrupt affinity is just a possible explanation, but if you're not modifying the interrupts in any non-standard way then it doesn't seem to be related.

It seems like the watchdog timeout command was triggered because the driver doesn't poll for the queue's completions. This *seems* to happen because the napi function (the networking bottom-half) isn't being scheduled by the CPU to run.
There might be two explanations to why this can happen:
- some race in the kernel caused the softIRQ scheduling to be missed. This possibility is what led me to ask about your affinity setup. After internal testing, I couldn't reproduce such issue
- the kernel doesn't have a free timeslot to give to the softIRQ handler. Given the high priority of this handler (part of the irqd process), there shouldn't be userspace programs capable of depriving napi from running **unless given an unusually high priority by superuser**. Kernel processes, on the other hand, should be able to starve napi, but this also shouldn't happen for normal flow.

I'd like to take a look at the second possibility. Is it possible for you to fetch the CPU statistics for the instance in one sec intervals continuously ? I'd need an output of a command like
```
mpstat -P ALL 1
```

which should cover the time of the reset. This output could tell what kept each CPU busy at the time of the driver reset (note that it doesn't reveal any process information other than the classification of the CPU time into kernel-space, userspace, softIRQ etc.).
If the issue is indeed a lack of timeslots for the napi handler, I would expect to see the command's output to show that the CPU passes all its time in kernel space.

p.s.

If you actually use superuser rights to allow userspace programs run in realtime priority (or one level beneath it) please let me know

PaulFurtado, 2025-03-18:
I will work on getting the mpstat data, it's just going to require some engineering though because it outputs quite a bit every second so we cannot just let it go into our normal log pipelines. I will have to run it on all of our kubernetes nodes with separate log rotation and figure something out so that we get the last several minutes of logs off of the machine when it hits the ENA issue before the machine gets killed.

> If you actually use superuser rights to allow userspace programs run in realtime priority (or one level beneath it) please let me know

I don't believe that we do anywhere, but it is possible that something is somewhere doing so unexpectedly. I will do some scanning to see what I find.

Do you think that EBS attachment/detachment may be the issue? Our kube nodes have a high density of pods using persistent volumes, so in the realtime process list on the nodes I spot checked, there are a ton of threads for each EBS nvme device. Example list of realtime processes (which are all kernel threads): https://git.hubteam.com/gist/pfurtado/73ffe8811a6a1fc40726df976bee50b2

ShayAgros, 2025-03-19:
> I will work on getting the mpstat data, it's just going to require some engineering though because it outputs quite a bit every second so we cannot just let it go into our normal log pipelines

You could do something like creating a logstream and piping mpstats output to it. It'd probably stop sending new stats once the interface starts disconnecting but it'd give us more visibility to what happened before. Something like the following Cloudwatch agent configurations (thank god for AI because I could spend the whole day reading wikis to achieve this):
```
{
  "agent": {
    "metrics_collection_interval": 1,
    "run_as_user": "root"
  },
  "logs": {
    "logs_collected": {
      "files": {
        "collect_list": [
          {
            "file_path": "/home/ec2-user/my_mpstats.txt",
            "log_group_name": "/aws/mpstats",
            "log_stream_name": "stats",
            "timestamp_format": "%Y-%m-%d %H:%M:%S"
          }
        ]
      }
    }
  }
}
```

and then have a simple script running like
```
while true; do
mpstat -P ALL -o JSON | xargs >> my_mpstats.txt
sleep 1
done
```

This can also be just an S3 operation that sends logs to a bucket.

> so in the realtime process list on the nodes I spot checked, there are a ton of threads for each EBS nvme device

This caught my eye and I tried to make an experiment in which I spawn like 5000 processes which spin in a while loop with a maximum priority of a FIFO scheduling. (see [this gist](https://gist.github.com/ShayAgros/c04322f6ed7578ab24260b9dc10f8835)). 

I scheduled all process on CPU 0 along with a program which prints (or rather tries) a message once a second to see how much the realtime processes are able to starve the CPU.

In addition I ran an iperf UDP stream which has its affinity, XPS, and CPU set to core 0, queue 0.
At peak points I managed to starve the normal printing program for 12-13 seconds, but I didn't manage to starve napi \\=

I'm (very very) far from being Linux scheduling expert, so I might have done things wrong, but if the EBS realtime processes are indeed what prevents napi from running, then it might be done through kernel space.

olyazavr, 2025-03-31:
I have two samples of mpstat data from two different hosts, taken every second, starting from several minutes before the hosts had an ENA driver reset until the reset:
https://gist.github.com/olyazavr/ae46c0a7315e96580a516d8bf9b2a498
https://gist.github.com/olyazavr/8e440fa7c1005dba66d2714c32ba8125

ShayAgros, 2025-04-03:
Hi Olga, thanks for sharing it. Can you please give me the instance ids and region ?

It would allow me to correlate the mpstats results with the reset

olyazavr, 2025-04-03:
The first one (ip-172-18-82-184.ec2.internal) is i-05144ca77bafd3229. The second (ip-172-18-65-61.ec2.internal) is i-07c1e144193d42f25. Both are in us-east-1

davidarinzon, 2025-04-24:
Hi @PaulFurtado / @olyazavr 

As mentioned previously, we recommend following the path of opening a ticket with AWS support and continuing the analysis and discussions there. The relevant teams will be pulled in order to look into the issue.
You may reference this github ticket in the request in order to provide more information and context.