amzn/amzn-drivers #337: [Support]:

mtwhitecamel, 2025-03-12:
### Preliminary Actions

- [x] I have searched the existing [issues](https://github.com/amzn/amzn-drivers/issues) and didn't find a duplicate.
- [x] I have followed the [AWS official troubleshoot documentation](https://docs.aws.amazon.com/AWSEC2/latest/UserGuide/troubleshooting-ena.html).
- [x] I have followed the driver [readme](https://github.com/amzn/amzn-drivers) and [best practices](https://github.com/amzn/amzn-drivers/blob/master/kernel/linux/ena/ENA_Linux_Best_Practices.rst).

### Driver Type

DPDK PMD for Elastic Network Adapter (ENA)

### Driver Tag/Commit

dpdk:20.11.9

### Custom Code

No

### OS Platform and Distribution

Amazon Linux 2

### Support request

driver_version: igb_uio from dpdk-kmods:latest(9b182b)
NIC: Elastic Network Adapter (ENA) ec20

I want to receive multicast by dpdk::rte_eth_dev_set_mc_addr_list() but the function returned errorcode ENOTSUP(-95).
Also tested several related function: rte_eth_allmulticast_enable(), rte_eth_promiscuous_enable(), rte_eth_dev_mac_addr_add(), and all of them returned ENOTSUP.
Tried to bind NIC with vfio-pci driver doesn't work neither, is that means ENA doesn't support dpdk multicast feature?

### Contact Details

mtwhitecamel@gmail.com

mtwhitecamel, 2025-03-12:
https://github.com/amzn/amzn-drivers/tree/master/userspace/dpdk  I cannot find anything about multicast from this document, hope I didn't miss something important.

shaibran, 2025-03-16:
Hello [mtwhitecamel](https://github.com/mtwhitecamel),

The mentioned APIs are indeed unsupported and will remain so, as they cannot coexist with EC2 architecture. 
You can configure multicast through [TGW](https://docs.aws.amazon.com/vpc/latest/tgw/tgw-multicast-overview.html).

All the best,
Shai

shaibran, 2025-03-18:
I am resolving this support issues, feel free to reopen in case you have any follow up questions

closed 2025-03-18 by shaibran

mtwhitecamel, 2025-03-27:
got it, thanks bro