amzn/amzn-drivers #344: [Bug]: Compilation Issue on AWS Linux 1 / Kernel 4.4.35-33.55

AaronKalair, 2025-05-02:
### Preliminary Actions

- [x] I have searched the existing [issues](https://github.com/amzn/amzn-drivers/issues) and didn't find a duplicate.
- [x] I have followed the [AWS official troubleshoot documentation](https://docs.aws.amazon.com/AWSEC2/latest/UserGuide/troubleshooting-ena.html).
- [x] I have followed the driver [readme](https://github.com/amzn/amzn-drivers) and [best practices](https://github.com/amzn/amzn-drivers/blob/master/kernel/linux/ena/ENA_Linux_Best_Practices.rst).

### Driver Type

Linux kernel driver for Elastic Network Adapter (ENA)

### Driver Tag/Commit

ena_linux_2.14.0

### Custom Code

No

### OS Platform and Distribution

Linux <HOSTNAME REDACTED> 4.4.35-33.55.amzn1.x86_64 #1 SMP Tue Dec 6 20:30:04 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux

### Bug description

Hi,

We're trying to install the latest version of the ENA Driver on AWS Linux 1 so we can get access to the [Performance Metrics](https://docs.aws.amazon.com/AWSEC2/latest/UserGuide/monitoring-network-performance-ena.html)

We updated our kernel version to the latest available but that only gets us to ENA Version 1.1.1

```
modinfo ena
filename:       /lib/modules/4.4.35-33.55.amzn1.x86_64/kernel/drivers/amazon/net/ena/ena.ko
version:        1.1.1
license:        GPL
description:    Elastic Network Adapter (ENA)
author:         Amazon.com, Inc. or its affiliates
srcversion:     4419A5D8E02CB50FF7C4AC0
alias:          pci:v00001D0Fd0000EC21sv*sd*bc*sc*i*
alias:          pci:v00001D0Fd0000EC20sv*sd*bc*sc*i*
alias:          pci:v00001D0Fd00001EC2sv*sd*bc*sc*i*
alias:          pci:v00001D0Fd00000EC2sv*sd*bc*sc*i*
depends:
intree:         Y
vermagic:       4.4.35-33.55.amzn1.x86_64 SMP mod_unload modversions
parm:           debug:Debug level (0=none,...,16=all) (int)
```

So I'm following the [instructions](https://github.com/amzn/amzn-drivers/tree/master/kernel/linux/ena#driver-installation-using-dkms) to compile it from scratch using DKMS

When I run `sudo dkms build -m amzn-drivers -v 2.14.0` it fails

```CC: /usr/bin/gcc48

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
make KERNELRELEASE=4.4.35-33.55.amzn1.x86_64 -C kernel/linux/ena/ BUILD_KERNEL=4.4.35-33.55.amzn1.x86_64...(bad exit status: 2)
Error! Bad return status for module build on kernel: 4.4.35-33.55.amzn1.x86_64 (x86_64)
Consult /var/lib/dkms/amzn-drivers/2.14.0/build/make.log for more information.
```

`make.log` has this error

```
### Detection completed ###
make -C /lib/modules/4.4.35-33.55.amzn1.x86_64/build M=/var/lib/dkms/amzn-drivers/2.14.0/build/kernel/linux/ena modules
make[1]: Entering directory `/usr/src/kernels/4.4.35-33.55.amzn1.x86_64'
  CC [M]  /var/lib/dkms/amzn-drivers/2.14.0/build/kernel/linux/ena/ena_netdev.o
In file included from /var/lib/dkms/amzn-drivers/2.14.0/build/kernel/linux/ena/ena_netdev.h:9:0,
                 from /var/lib/dkms/amzn-drivers/2.14.0/build/kernel/linux/ena/ena_netdev.c:27:
/var/lib/dkms/amzn-drivers/2.14.0/build/kernel/linux/ena/kcompat.h: In function ‘ena_dma_unmap_page_attrs’:
/var/lib/dkms/amzn-drivers/2.14.0/build/kernel/linux/ena/kcompat.h:1085:2: error: implicit declaration of function ‘dma_unmap_page_attrs’ [-Werror=implicit-function-declaration]
  dma_unmap_page_attrs(dev, addr, size, dir, attrs);
  ^
compilation terminated due to -Wfatal-errors.
cc1: some warnings being treated as errors
make[2]: *** [/var/lib/dkms/amzn-drivers/2.14.0/build/kernel/linux/ena/ena_netdev.o] Error 1
make[1]: *** [_module_/var/lib/dkms/amzn-drivers/2.14.0/build/kernel/linux/ena] Error 2
make[1]: Leaving directory `/usr/src/kernels/4.4.35-33.55.amzn1.x86_64'
make: *** [ena.ko] Error 2
make: Leaving directory `/var/lib/dkms/amzn-drivers/2.14.0/build/kernel/linux/ena'
```

I know AWS Linux 1 is very old but I see it listed as supported in your [documentation](https://github.com/amzn/amzn-drivers/blob/master/kernel/linux/ena/RELEASENOTES.md#supported-kernel-versions-and-distributions)

### Reproduction steps

```bash
1. sudo yum install dkms
2. sudo yum install git
3. git clone https://github.com/amzn/amzn-drivers.git
4. sudo mv amzn-drivers /usr/src/amzn-drivers-2.14.0
5. sudo vi /usr/src/amzn-drivers-2.14.0/dkms.conf

PACKAGE_NAME="ena"
PACKAGE_VERSION="1.0.0"
CLEAN="make -C kernel/linux/ena clean"
MAKE="make -C kernel/linux/ena/ BUILD_KERNEL=${kernelver}"
BUILT_MODULE_NAME[0]="ena"
BUILT_MODULE_LOCATION="kernel/linux/ena"
DEST_MODULE_LOCATION[0]="/updates"
DEST_MODULE_NAME[0]="ena"
REMAKE_INITRD="yes"
AUTOINSTALL="yes"

7. sudo dkms add -m amzn-drivers -v 2.14.0
8. sudo dkms build -m amzn-drivers -v 2.14.0
```

### Expected Behavior

The compliation to complete succesfully

### Actual Behavior

I got a compile error

```
### Detection completed ###
make -C /lib/modules/4.4.35-33.55.amzn1.x86_64/build M=/var/lib/dkms/amzn-drivers/2.14.0/build/kernel/linux/ena modules
make[1]: Entering directory `/usr/src/kernels/4.4.35-33.55.amzn1.x86_64'
  CC [M]  /var/lib/dkms/amzn-drivers/2.14.0/build/kernel/linux/ena/ena_netdev.o
In file included from /var/lib/dkms/amzn-drivers/2.14.0/build/kernel/linux/ena/ena_netdev.h:9:0,
                 from /var/lib/dkms/amzn-drivers/2.14.0/build/kernel/linux/ena/ena_netdev.c:27:
/var/lib/dkms/amzn-drivers/2.14.0/build/kernel/linux/ena/kcompat.h: In function ‘ena_dma_unmap_page_attrs’:
/var/lib/dkms/amzn-drivers/2.14.0/build/kernel/linux/ena/kcompat.h:1085:2: error: implicit declaration of function ‘dma_unmap_page_attrs’ [-Werror=implicit-function-declaration]
  dma_unmap_page_attrs(dev, addr, size, dir, attrs);
  ^
compilation terminated due to -Wfatal-errors.
cc1: some warnings being treated as errors
make[2]: *** [/var/lib/dkms/amzn-drivers/2.14.0/build/kernel/linux/ena/ena_netdev.o] Error 1
make[1]: *** [_module_/var/lib/dkms/amzn-drivers/2.14.0/build/kernel/linux/ena] Error 2
make[1]: Leaving directory `/usr/src/kernels/4.4.35-33.55.amzn1.x86_64'
make: *** [ena.ko] Error 2
make: Leaving directory `/var/lib/dkms/amzn-drivers/2.14.0/build/kernel/linux/ena'
```

### Additional Data

```bash

```

### Relevant log output

```shell

```

### Contact Details

aaron.kalair@hudl.com

davidarinzon, 2025-05-04:
Hi @AaronKalair 

Thanks for pointing this issue out.
This API was introduced in Linux Kernel 4.10.
The AMI that you're using is fairly old and challenging to acquire, it is also EOL for several years, therefore, it makes sense that you can't get an updated driver with the AMI update.

As you're interested in the performance metrics, I can offer the following two options:

1) A point fix for this issue
2) A recommendation to use driver release 2.8.6 (https://github.com/amzn/amzn-drivers/releases/tag/ena_linux_2.8.6) which has the features that you require and doesn't have this API introduced. To be on the safer side, you can try 2.8.1 (https://github.com/amzn/amzn-drivers/releases/tag/ena_linux_2.8.1) which includes the introduction of the performance metrics.

Please let us know what works for you.

AaronKalair, 2025-05-09:
Hey

Thanks for the detailed reply.

I tried version 2.8.6 but got the same compilation errors

Version 2.8.1 compiled and works fine though, exposing the ENA Metrics we want

Thanks for your help!

closed 2025-06-08 by davidarinzon