amzn/amzn-drivers #345: [Feature request]: AF_XDP with XDP_ZEROCOPY?

dforsber, 2025-05-07:
### Preliminary Actions

- [x] I have searched the existing [issues](https://github.com/amzn/amzn-drivers/issues) and didn't find a duplicate.

### Driver Type

Linux kernel driver for Elastic Network Adapter (ENA)

### Motivation

I've read various documentation and the assumption is that AWS ENA supports AF_XDP sockets with XDP_ZEROCOPY. I've set the channels to half of the max, configured MTU etc. But regardless of huge tables or not, aligned memory, and even Linux distro (AL2, Ubuntu) and several instance types tested. My code works only with XDP_COPY and not XDP_ZEROCOPY.

Take for example this library, and the "dump" example in it:
https://github.com/aterlo/afxdp-rs

```shell
ubuntu@ip-172-31-3-217:~/afxdp-rs$ date
Wed May  7 12:45:51 PM UTC 2025
ubuntu@ip-172-31-3-217:~/afxdp-rs$ uname -a
Linux ip-172-31-3-217 6.8.0-1024-aws #26-Ubuntu SMP Tue Feb 18 17:22:37 UTC 2025 x86_64 x86_64 x86_64 GNU/Linux
ubuntu@ip-172-31-3-217:~/afxdp-rs$ sudo ./target/release/examples/dump --huge-tlb --zero-copy --link-name lo
Total buffers=65536 buffer size=2048
^C

thread 'main' panicked at examples/dump.rs:137:21:
no socket for you: Create(Os { code: 95, kind: Uncategorized, message: "Operation not supported" })
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
ubuntu@ip-172-31-3-217:~/afxdp-rs$ sudo ./target/release/examples/dump --huge-tlb --link-name lo
Total buffers=65536 buffer size=2048
Adding 2048 buffers to the fill queue

root@ip-172-31-3-217:/home/ubuntu/afxdp-rs# ./target/release/examples/dump --zero-copy --link-name ens5
Total buffers=65536 buffer size=2048
thread 'main' panicked at examples/dump.rs:137:21:
no socket for you: Create(Os { code: 95, kind: Uncategorized, message: "Operation not supported" })
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
root@ip-172-31-3-217:/home/ubuntu/afxdp-rs# 
root@ip-172-31-3-217:/home/ubuntu/afxdp-rs# 
root@ip-172-31-3-217:/home/ubuntu/afxdp-rs# 
root@ip-172-31-3-217:/home/ubuntu/afxdp-rs# ./target/release/examples/dump --link-name ens5
Total buffers=65536 buffer size=2048
Adding 2048 buffers to the fill queue
Some(EthernetPacket { destination : 02:23:58:09:6a:4f, source : 02:de:4a:42:f1:8d, ethertype : EtherType(2048),  })
Some(EthernetPacket { destination : 02:23:58:09:6a:4f, source : 02:de:4a:42:f1:8d, ethertype : EtherType(2048),  })

```

### Solution

I'm probably doing something wrong as this feature should be enabled already?

### Contact Details

dforsber@gmail.com

davidarinzon, 2025-05-07:
Hi @dforsber 

Thanks for reaching out.
AF_XDP with ZC is supported from 2.13.0g version, given that you've mentioned AL2, I assume that you're using a later driver (Wasn't mentioned in the ticket).

- Do you observe any errors in the dmesg?
- Have you tried simpler programs such as https://github.com/xdp-project/bpf-examples/tree/main/AF_XDP-example?

davidarinzon, 2025-05-07:
@dforsber 

If you want AF_XDP support, you need to take the ENA github driver starting from 2.13.0g, as I previously noted.

Most distributions don't show the driver version, it may be either 2.1.0K or the version of the kernel itself.
The ENA driver in the Linux Kernel doesn't support AF_XDP.

Please make sure you're using the relevant driver and let us know if you face this issue.
You've mentioned AL2, which contains the ENA github driver, but without mentioning the version.

I would recommend loading the latest ENA github driver on the Ubuntu that you're using.

dforsber, 2025-05-08:
Ah, got it. Thank you very much, it works now 👍🏻