amzn/amzn-drivers #346: [Bug]: ena_com_get_feature response is accessed without checking return value first

gurugubs, 2025-06-02:
### Preliminary Actions

- [x] I have searched the existing [issues](https://github.com/amzn/amzn-drivers/issues) and didn't find a duplicate.
- [x] I have followed the [AWS official troubleshoot documentation](https://docs.aws.amazon.com/AWSEC2/latest/UserGuide/troubleshooting-ena.html).
- [x] I have followed the driver [readme](https://github.com/amzn/amzn-drivers) and [best practices](https://github.com/amzn/amzn-drivers/blob/master/kernel/linux/ena/ENA_Linux_Best_Practices.rst).

### Driver Type

Linux kernel driver for Elastic Network Adapter (ENA)

### Driver Tag/Commit

ena_linux_2.14.1

### Custom Code

No

### OS Platform and Distribution

All versions of Linux.

### Bug description

We are copying from response of ena_com_get_feature before checking for return value: https://github.com/amzn/amzn-drivers/blame/master/kernel/linux/common/ena_com/ena_com.c#L2246 - the actual check for return value is a couple of lines down that needs to be moved up.

This is a minor defect - it is very improbably to hit this case, but we should tidy up the code anyway. 

### Reproduction steps

```bash
Bug was found using a private emulator of the device for testing purposes.  Once you see the code, it is obvious that the defect exists and is trivial - no reproduction necessary.
```

### Expected Behavior

Return value of ena_com_get_feature must be checked before copying items out of the response structure.

### Actual Behavior

Return value is not checked, resulting in bogus data getting copied - this function eventually returns error, but we now copied something that should not have been.

### Additional Data

```bash

```

### Relevant log output

```shell

```

akiyano, 2025-06-02:
Thanks for submitting this bug report, we will look into it and update here when it is fixed.

davidarinzon, 2025-06-19:
Hello
Thanks for uploading this pull request
We've released the changes today, the specific patch is https://github.com/amzn/amzn-drivers/commit/08cde87d70cbf23d03e6d437167e07cc4b237f28