amzn/amzn-drivers #348: [Bug]: ENA VFIO patch not working on Amazon Linux 2023

selevit, 2025-07-03:
### Preliminary Actions

- [x] I have searched the existing [issues](https://github.com/amzn/amzn-drivers/issues) and didn't find a duplicate.
- [x] I have followed the [AWS official troubleshoot documentation](https://docs.aws.amazon.com/AWSEC2/latest/UserGuide/troubleshooting-ena.html).
- [x] I have followed the driver [readme](https://github.com/amzn/amzn-drivers) and [best practices](https://github.com/amzn/amzn-drivers/blob/master/kernel/linux/ena/ENA_Linux_Best_Practices.rst).

### Driver Type

DPDK PMD for Elastic Network Adapter (ENA)

### Driver Tag/Commit

ena_linux_2.13.1

### Custom Code

No

### OS Platform and Distribution

Linux dpdk-pg1 6.1.115-126.197.amzn2023.x86_64 #1 SMP PREEMPT_DYNAMIC Tue Nov  5 17:36:57 UTC 2024 x86_64 x86_64 x86_64 GNU/Linux

### Bug description

Write combining doesn't work on AWS Linux 2023 with vfio-pci patched via [get-vfio-with-wc.sh](https://github.com/amzn/amzn-drivers/blob/296d60c4ed18cac5edf3311124d39777fedb94a2/userspace/dpdk/enav2-vfio-patch/get-vfio-with-wc.sh).

### Reproduction steps

```bash
# Install write-combined patch
bash get-vfio-with-wc.sh

# Reboot just in case, to 100% unload all old modules
reboot 

# Enable NOIOMMU mode and load dpdk driver
echo 1 > /sys/module/vfio/parameters/enable_unsafe_noiommu_mode
dpdk-devbind.py --bind=vfio-pci 00:06.0

# Check prefetchable base address
➜ lspci -v | grep -A 10 '00:06.0'
00:06.0 Ethernet controller: Amazon.com, Inc. Elastic Network Adapter (ENA)
	Subsystem: Amazon.com, Inc. Elastic Network Adapter (ENA)
	Physical Slot: 6
	Flags: fast devsel, IOMMU group 0
	Memory at c0606000 (32-bit, non-prefetchable) [size=8K]
	Memory at c0604000 (32-bit, non-prefetchable) [size=8K]
	Memory at c0500000 (32-bit, prefetchable) [size=1M]
	Capabilities: [70] Express Endpoint, MSI 00
	Capabilities: [b0] MSI-X: Enable- Count=12 Masked-
	Kernel driver in use: vfio-pci
	Kernel modules: ena

./bin/dpdk-testpmd

# Then check Write Combining

➜ cat /sys/kernel/debug/x86/pat_memtype_list  | grep c0500000
PAT: [mem 0x00000000c0400000-0x00000000c0500000] write-combining
PAT: [mem 0x00000000c0500000-0x00000000c0600000] uncached-minus

I also tried running `./bin/dpdk-testpmd -a "06:00.0,llq_policy=2"` to force enable LLQ mode, but it didn't help.

Everything works fine with `igb_uio`. However, I need some `vfio-pci` specific features, that's why `igb_uio` isn't an option.
```

### Expected Behavior

It should show write-combining for the both lines in this output

```
cat /sys/kernel/debug/x86/pat_memtype_list  | grep c0500000
PAT: [mem 0x00000000c0400000-0x00000000c0500000] write-combining
PAT: [mem 0x00000000c0500000-0x00000000c0600000] uncached-minus
```

### Actual Behavior

It shows `uncached-minus` for the addresses started from `c0500000`.

### Additional Data

```bash

```

### Relevant log output

```shell

```

amitbern-aws, 2025-07-06:
Thanks for informing us, we will look into this.

amitbern-aws, 2025-07-06:
@selevit 
I tested the write-combine configuration on a c6in.16xlarge instance using Amazon Linux across multiple kernels in the eu-central-1 region:
- Your AMI: ami-0b5673b5f6e8f7fa7 (AL2023 version 2023.6.20241121.0, kernel 6.1.115-126.197.amzn2023.x86_64)
- The latest AMI: ami-0229b8f55e5178b65 (AL2023 version 2023.7.20250623.1, kernel 6.1.141-155.222.amzn2023.x86_64)

In both cases, the configuration worked as expected.

Notice:
1. you need to load the vfio driver after running get-vfio-with-wc.sh (no need for reboot)
2. you need the test-pmd to run in parallel while you check the BAR2 memory address for write combine configuration

Steps:

```
# Create vfio driver with write combine
sudo yum install kernel-devel-$(uname -r) git -y
git clone https://github.com/amzn/amzn-drivers
cd amzn-drivers/userspace/dpdk/enav2-vfio-patch/
sudo ./get-vfio-with-wc.sh
sudo modprobe vfio-pci

# Install prerequsits
sudo yum install -y libatomic git kernel-devel kernel-headers numactl numactl-libs numactl-devel gcc python3 python-pip
sudo pip3 install meson ninja
sudo pip3 install pyelftools
cd ../../../..

# Enable NOIOMMU mode
sudo su
echo 1 > /sys/module/vfio/parameters/enable_unsafe_noiommu_mode
exit

# Add hugepages
echo 4096  | sudo tee /proc/sys/vm/nr_hugepages

# Bind second ENI
sudo ./usertools/dpdk-devbind.py --unbind 00:06.0 --force
sudo ./usertools/dpdk-devbind.py --bind=vfio-pci 00:06.0

# Check BAR2 write combine configuration
sudo lspci -v | grep -A 10 '00:06.0'
00:06.0 Ethernet controller: Amazon.com, Inc. Elastic Network Adapter (ENA)
        Subsystem: Amazon.com, Inc. Elastic Network Adapter (ENA)
        Physical Slot: 6
        Flags: fast devsel, IOMMU group 0
        Memory at c0608000 (32-bit, non-prefetchable) [size=8K]
        Memory at c060a000 (32-bit, non-prefetchable) [size=8K]
        Memory at c0800000 (32-bit, prefetchable) [size=2M]
        Capabilities: [70] Express Endpoint, MSI 00
        Capabilities: [b0] MSI-X: Enable- Count=20 Masked-
        Kernel driver in use: vfio-pci
        Kernel modules: ena
        
sudo cat /sys/kernel/debug/x86/pat_memtype_list | grep c0800000
<empty>

 # Compile and run testpmd on another ssh connection
git clone https://github.com/DPDK/dpdk-stable.git
cd dpdk-stable/
git checkout v23.11 (for ami ami-0b5673b5f6e8f7fa7)
git checkout v24.11.2 (for ami ami-0229b8f55e5178b65)
meson build
ninja -C build
sudo ./build/app/dpdk-testpmd -- -i

# Check BAR2 write combine configuration again
sudo cat /sys/kernel/debug/x86/pat_memtype_list | grep c0800000
PAT: [mem 0x00000000c0800000-0x00000000c0a00000] write-combining
```

Tell me if you still have the same issue after steps above.

Thanks

selevit, 2025-07-07:
@amitbern-aws, I've just executed exactly the same commands as you provided above, prematurely unloading all vfio-pci related modules, and it didn't work on my AMI with both DPDK versions that you suggested above (v24.11.2 and v23.11).

That's what I see when the `testpmd` app is not running:

```
sudo cat /sys/kernel/debug/x86/pat_memtype_list | grep c0500000
PAT: [mem 0x00000000c0400000-0x00000000c0500000] write-combining
```

And that's what I see when It is running:

```
sudo cat /sys/kernel/debug/x86/pat_memtype_list | grep c0500000
PAT: [mem 0x00000000c0400000-0x00000000c0500000] write-combining
PAT: [mem 0x00000000c0500000-0x00000000c0600000] uncached-minus
```

If needed, I can provide you the access to this particular instance - this is a playground one. If you are ok with this, pls send me an email with your SSH public key to the email address in my profile.

amitbern-aws, 2025-07-07:
Hi,
Sent further debug request to your email
Thanks

amitbern-aws, 2025-07-08:
@selevit 
Thanks for checking.
I'll close this issue, and email regarding older AMI once we'll have some answers.
Thanks

closed 2025-07-09 by shaibran